makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #1: can't compile it

AndyCGN, 2020-04-17:
I can't compile it, to many errors even without any change to the configuration.h file. Is there something to install additionally bevor compiling?

Kokorone, 2020-04-18:
I found that installing lvgl and lvgl drivers in platformio gets you further than without, but I'm still having compiling hang at the last step. I also had to define some pins that were not defined.

makerbase-mks, 2020-04-18:
@Kokorone Can you show the compile information?

AndyCGN, 2020-04-18:
Let me try your pin enhancements first. I guess that will fix a lot of errors in my case too. But I also run in the issue with libU8glib-HAL_ID1932.a

Kokorone, 2020-04-18:
> @Kokorone Can you show the compile information?

Please refer to my issue found here 

https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/issues/2

MKS-Sean, 2020-04-21:
We have made a fix, you can update and try it.

makerbase-mks, 2020-04-22:
Fixed

closed 2020-04-22 by makerbase-mks