makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #2: Had to define several pins and values manually, but compiling hangs at libU8glib-HAL_ID1932.a

Kokorone, 2020-04-17:
[image]

Attached are the configuration.h, configuration_adv.h, and pins file for the robin nano. I have it configured for use in the Creativity3D ELF printer using BLTouch, but there is some issue with the TFT35 code that makerbase has inserted that causes the firmware to hang once it gets to libU8glib-HAL_ID1932.a

Please review and update, I'd like to use this firmware.
[Firmware.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/4495502/Firmware.zip)

Kokorone, 2020-04-18:
In this link is the entire firmware folder I am using to compile, I have only changed a few pins and the configuration files. Please try and compile so you can see where it is having issues.

[https://mega.nz/file/HFty0QaS#MgQHLEU0O8oLYONiuKznQe4azddxTo_O2_SuiOj5_v4](url)

makerbase-mks, 2020-04-18:
It seems to be a problem relative with the library,  not with the firmware itself. We are checking it .

chenxiaowei107, 2020-04-20:
I have the same problem

MKS-Sean, 2020-04-21:
We have made a fix, you can update and try it.

Kokorone, 2020-04-21:
> We have made a fix, you can update and try it.

Getting a new error regarding lv_conf.h being missing.

[image]

MKS-Sean, 2020-04-21:
I can compile it well. Did you change something for it?

makerbase-mks, 2020-04-21:
@Kokorone, it is strange that your file doesn't match the one on github:
[image]
Maybe you can try to delete all the files and folders below ".pio" before compile again.

chenxiaowei107, 2020-04-21:
能编译成功了，但是刷进去以后黑屏不能开机。。

Kokorone, 2020-04-21:
> @Kokorone, it is strange that your file doesn't match the one on github:
> [image]
> Maybe you can try to delete all the files and folders below ".pio" before compile again.

Strange, but I had to wipe my .platformio lib folder. lvgl must have had a cached entry causing issues. Compiling now

closed 2020-04-21 by Kokorone

makerbase-mks, 2020-04-22:
@Kokorone Solved?

Kokorone, 2020-04-22:
> @Kokorone Solved?

yes, it builds now. Thank you