makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #6: If define EEPROM,it will not work

chenxiaowei107, 2020-04-22:
if I define EEPROM , the firmware will not work.

but I google It, if I add this code in pins_MKS_ROBIN_NANO.h ,It will work,
I Don't what does this code mean，so，If it's right please fix It

#define SDCARD_CONNECTION ONBOARD
#define SDCARD_EEPROM_EMULATION

Also the Bltouch pin is not defin in the file ,please fix it.

#define SERVO0_PIN         PA8

Kokorone, 2020-04-22:
Just submitted a PR to bring the pins file in line with vanilla marlin, which already has SDCARD_CONNECTION ONBOARD and #define SDCARD_EEPROM_EMULATION in there.
https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/pull/8

Also regarding the bltouch pin not being defined, it should be already merged here:
https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/commit/719e72e544d5d7f63207617e19b69c573400f81f

Miraculix200, 2020-04-22:
That is a bug in Marlin. It will be fixed in the next version of Marlin.

See https://github.com/MarlinFirmware/Marlin/issues/17376

MKS-Sean, 2020-04-23:
We merged Kokorone's Pull requests, which has reduced the clock speed of SDIO and fixed the problem that it cannot start after enabling eeprom.

chenxiaowei107, 2020-04-23:
> We merged Kokorone's Pull requests, which has reduced the clock speed of SDIO and fixed the problem that it cannot start after enabling eeprom.

I define Z_STEPPER_AUTO_ALIGN in the adv.h,but no menu in the screen ,why?

MKS-Sean, 2020-04-23:
@chenxiaowei107 当前的是一个基础版本，后续会持续添加.

MKS-Sean, 2020-04-23:
The current version is a basic version and will continue to be added in the future.

chenxiaowei107, 2020-04-23:
我想利用wifi的针脚连接octoprint，但是

closed 2020-04-23 by chenxiaowei107