makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #7: Printer Halted!

fenoloji, 2020-04-22:
My printer has error: "Printer halted"

Miraculix200, 2020-04-22:
What did you do before the error?

fenoloji, 2020-04-22:
> What did you do before the error?

After upgrading firmware.

fenoloji, 2020-04-22:
ompiling .pio\build\mks_robin_nano35\src\src\gcode\bedlevel\G42.cpp.o
In file included from Marlin\src\feature\touch\../../inc/../HAL/./STM32F1/HAL.h:36:0,
                 from Marlin\src\feature\touch\../../inc/../HAL/HAL.h:26,
                 from Marlin\src\feature\touch\../../inc/MarlinConfig.h:30,
                 from Marlin\src\feature\touch\xpt2046.cpp:25:
Marlin\src\feature\touch\xpt2046.cpp: In static member function 'static void XPT2046::init()':
Marlin\src\feature\touch\xpt2046.cpp:40:26: error: 'CS_PIN' was not declared in this scope
   #define TOUCH_CS_PIN   CS_PIN
                          ^
Marlin\src\feature\touch\../../inc/../HAL/./STM32F1/fastio.h:37:55: note: in definition of macro '_SET_MODE'
 #define _SET_MODE(IO,M)         gpio_set_mode(PIN_MAP[IO].gpio_device, PIN_MAP[IO].gpio_bit, M)
                                                       ^~
Marlin\src\feature\touch\../../inc/../HAL/./STM32F1/fastio.h:41:37: note: in expansion of macro '_SET_OUTPUT'
 #define OUT_WRITE(IO,V)         do{ _SET_OUTPUT(IO); WRITE(IO,V); }while(0)
                                     ^~~~~~~~~~~
Marlin\src\feature\touch\xpt2046.cpp:50:3: note: in expansion of macro 'OUT_WRITE'
   OUT_WRITE(TOUCH_CS_PIN, HIGH);
   ^~~~~~~~~
Marlin\src\feature\touch\xpt2046.cpp:50:13: note: in expansion of macro 'TOUCH_CS_PIN'
   OUT_WRITE(TOUCH_CS_PIN, HIGH);
             ^~~~~~~~~~~~
Marlin\src\feature\touch\xpt2046.cpp:40:26: note: suggested alternative: 'SS_PIN'
   #define TOUCH_CS_PIN   CS_PIN
                          ^
Marlin\src\feature\touch\../../inc/../HAL/./STM32F1/fastio.h:37:55: note: in definition of macro '_SET_MODE'
 #define _SET_MODE(IO,M)         gpio_set_mode(PIN_MAP[IO].gpio_device, PIN_MAP[IO].gpio_bit, M)
                                                       ^~
Marlin\src\feature\touch\../../inc/../HAL/./STM32F1/fastio.h:41:37: note: in expansion of macro '_SET_OUTPUT'
 #define OUT_WRITE(IO,V)         do{ _SET_OUTPUT(IO); WRITE(IO,V); }while(0)
                                     ^~~~~~~~~~~
Marlin\src\feature\touch\xpt2046.cpp:50:3: note: in expansion of macro 'OUT_WRITE'
   OUT_WRITE(TOUCH_CS_PIN, HIGH);
   ^~~~~~~~~
Marlin\src\feature\touch\xpt2046.cpp:50:13: note: in expansion of macro 'TOUCH_CS_PIN'
   OUT_WRITE(TOUCH_CS_PIN, HIGH);

Kokorone, 2020-04-22:
Sounds like its missing a pin definition for the touchmi probe. I don't have one to test, but I'd play around with that. Someone had a similar problem here, reading through it might help.

https://github.com/MarlinFirmware/Marlin/issues/15439

MKS-Sean, 2020-04-23:
Does the screen display other information? In addition to "Printer halted".

fenoloji, 2020-04-23:
The problem is resolved after removing the MKS PWC pins from motherboard. But X Y Z axis do not going home all.

Kokorone, 2020-04-24:
> The problem is resolved after removing the MKS PWC pins from motherboard. But X Y Z axis do not going home all.

I have disabled the pin definitions by default for the PWC and BLTouch, they only work on the v1.2 version of the nano. Thanks for catching that

https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/pull/11

MKS-Sean, 2020-04-24:
@Kokorone You are right, V1.2 adds BLTOUCH pin PA8 on the basis of V1.1, and changes the WI-IO0 pin from the original PA8 to PC13.
@fenoloji Is your motherboard version V1.1?

fenoloji, 2020-04-25:
> @Kokorone You are right, V1.2 adds BLTOUCH pin PA8 on the basis of V1.1, and changes the WI-IO0 pin from the original PA8 to PC13.
> @fenoloji Is your motherboard version V1.1?

My motherboard Robin Nano35 v1.2

Kokorone, 2020-04-25:
I've submitted a new PR to potentially fix your problem. https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/pull/13

makerbase-mks, 2020-04-29:
Fixed. [Add BLTouch and MKS-PWC pin definitions #4](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/pull/4)

closed 2020-04-29 by makerbase-mks