makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #10: The WIFI serial PIN Is not work

chenxiaowei107, 2020-04-23:
I Want use WIFI PIN (TX PA9 RX PA10) to connect the octoprint , but it's not work.
How should I do ?
[image]

MKS-Sean, 2020-04-24:
I think the serial port has not been configured yet.

chenxiaowei107, 2020-04-24:
> I think the serial port has not been configured yet.

yes，but I don't know how to config.

ahalekelly, 2020-04-27:
As described in  `Configuration.h`, serial port 3 on the Robin Nano is the USB, and port 1 is PA9/PA10, the Wifi connector.

chenxiaowei107, 2020-04-27:
SO I should Define

ahalekelly, 2020-04-27:
Yep it's working for me, see #12

mks-viva, 2020-11-17:
@ahalekelly 
Has this problem been solved?

ahalekelly, 2020-11-17:
Unfortunately not. I got ESP3D to be able to communicate with the printer, but I couldn't upload print jobs reliably. I haven't been able to get the MKS Wifi working yet, but other people have.

mks-viva, 2020-11-17:
@ahalekelly 
You use MKS Robin Nano V1.x?

mks-viva, 2020-12-11:
Has this problem been resolved?

closed 2021-02-21 by MKS-Sean