makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #9: LCD blank

race12001, 2020-04-23:
The LCD shows booting when first powering on then goes blank. I have been able to get the older marlin screen to work but not the GUI screen firmware.

race12001, 2020-04-23:
Forgot to mention I am using the MKS Robin TFT35 V1.0 screen that came in the printer.

chenxiaowei107, 2020-04-23:
you should look this 
https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/issues/3#issuecomment-617283606

race12001, 2020-04-23:
Gave that a go with no luck. Shows that everything updated and still boots to a blank screen.

MKS-Sean, 2020-04-23:
You can try to update the Robin_nano35.bin, mks_pic, mks_font files that we posted in the Firmware folder.
[image]

race12001, 2020-04-23:
Updated the files and still doing the same thing.   I also recompiled firmware and still no change. I did load that bin file in the firmware folder and seen the screen but it was the smoothieware look and was extremely slow.

imberest, 2020-04-23:
Hello @race12001 , i was facing exactly the same issue as you are. try downloading a fresh copy.
The looks of screen can be copied from another repository. I used [MKS-TFT](https://github.com/DenisShelema/MKS-TFT3.5-light-and-dark-themes) light one for 3.5. 
So basically, build your project with all the files already copied into places, grab config.txt onto sd card. Grab **mks_font** and **mks_pic** from **Firmware** folder and **Robin_nano35.bin** from **.pio\build\mks_robin_nano35**. 
Hope it makes sense and helps you

fenelon, 2020-05-13:
> You can try to update the Robin_nano35.bin, mks_pic, mks_font files that we posted in the Firmware folder.
> [image]

It doesn't seem to work on Robin Nano v.1.1. The flashing is every time stuck on bmp_dir.

closed 2021-02-21 by MKS-Sean