makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #12: MKS Wifi Support?

ahalekelly, 2020-04-24:
Is there support for the MKS WiFi module in this firmware? After enabling WIFISUPPORT and WEBSUPPORT, there's still no WiFi option on the printer menu. Or is it possible to flash ESP3D to the MKS WiFi module?

MKS-Sean, 2020-04-24:
@ahalekelly We will add the wifi function as soon as possible, and welcome more developers to join the development of this firmware.

Miraculix200, 2020-04-24:
I'm not from MKS, but as I have experience with ESP8266 I'm quite sure it's possible to flash ESP3D on the MKS WiFi module. There's an ESP-12S chip on the MKS WiFi module as far as I know. The ESP-12S has enough flash space (4MB) for ESP3D.

ahalekelly, 2020-04-26:
Thank you MKS-Sean, I'm looking forward to it.

In the meantime I'll try ESP3D, my concern was mostly with whether the MKS bootloader on the ESP8266 would be compatible.

ahalekelly, 2020-04-27:
Wow, got ESP3D working quite easily.

1. Renamed  `esp3d.ino ` to  `esp3d.ino.cpp ` and used Platform.io to build with default settings
2. Renamed ESP3D  `firmware.bin ` to  `MksWifi.bin ` and uploaded via MKS web interface.
Presumably, the SD card or pulling the module and using a serial cable would work too.
3. Changed Marlin serial ports for the Robin Nano to 3 (USB) and 1 (ESP)

That's it!

closed 2020-04-27 by ahalekelly

ahalekelly, 2020-04-27:
Nevermind, ESP3D doesn't have a Cura plugin like the MKS Wifi, and the file uploads are incredibly 
 slow.
[WirelessPrinting](https://github.com/probonopd/WirelessPrinting/) has a Cura plugin but seems to require an SD card attached to the ESP8266 and doesn't support sending files to the printer's SD card.
The MKS Wifi worked far better 🙁

chenxiaowei107, 2020-04-27:
> Nevermind, ESP3D doesn't have a Cura plugin like the MKS Wifi, and the file uploads are incredibly
> slow.
> [WirelessPrinting](https://github.com/probonopd/WirelessPrinting/) has a Cura plugin but [...]

Yes，the speed is too slow,so I think the octoprint is better,I have mks wifi card,bu if i transmission a large gcode file ,It's cost too much time.

Loreagle, 2020-07-05:
I think that with the new GUI the MksWifi should be supported with it's firmware. Also if it slow for me is very easy to use and taking 5 minutes for transmission is less annoying than Taking SD, connect to PC, move Gcode, remove SD.....