makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #17: Couple of bugs to report: freezing printer and printing wrong files

imberest, 2020-04-28:
Hello all,
I just wanted to report a few bugs I have noticed so far.

**Printer freezing when using TFT35 screen**
I have noticed if I move X-axis and press it 3 times fairly quickly so it goes for example 30mm it will freeze and TFT gets non-responsive. By the looks, it freezes board too.
Also, while printing, tried to turn on FAN, and before doing it it froze.

**TFT touch is wrong**
So another bug I have noticed is that sometimes the wrong place of TFT is pressed. For example, I press to move Y-axis and Z or X moves instead.
Also, sometimes when selecting a file to print, it prints different file from SD.

Anyone else has had something similar?

ahalekelly, 2020-04-29:
Can confirm I've had it freeze when trying to move quickly via the menu.

MKS-Sean, 2020-04-29:
Yes, I have also tested this problem, and I am thinking of ways to fix it.

imberest, 2020-05-09:
@MKS-Sean Any luck? I found that they have **queue.enqueue_one_now(PSTR(public_buf_l));** in common. Could it be something with this queue?

closed 2021-02-21 by MKS-Sean