makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #18: FLASH_EEPROM_EMULATION doesn't Work, Bed Corner leveling and User Menus not shown

Alexoko12, 2020-04-28:
Hello,

1. I want to report a bug with EEPROM:
I have an v1.1 board and wanted to change in "Marlin\src\pins\stm32f1\pins_MKS_ROBIN_NANO.h" 

From:
//#define FLASH_EEPROM_EMULATION
#define SDCARD_EEPROM_EMULATION
to:
#define FLASH_EEPROM_EMULATION
//#define SDCARD_EEPROM_EMULATION

because I use Octoprint and don't want a SD Card just for EEPROM.DAT after Saving with M500 the touch stopped working and after reboot stays black.

2. I want to report a bug with the GUI:
a) I defined LEVEL_BED_CORNERS but on screen it isn't shown not under Tools nor under Settings. Is it possible with the GUI to activate this function?

b) Same Problem with User-defined menu. I updatet the screen with the pics and fonts folders from Firmware.
Cutout from "Marlin\Configuration.h"
// @section temperature

// Preheat Constants
//Alexoko12 #define PREHEAT_1_LABEL       "PLA" 
//Alexoko12 #define PREHEAT_1_TEMP_HOTEND 180 
//Alexoko12 #define PREHEAT_1_TEMP_BED     70 
//Alexoko12 #define PREHEAT_1_FAN_SPEED     0 // Value from 0 to 255

//Alexoko12 #define PREHEAT_2_LABEL       "ABS" 
//Alexoko12 #define PREHEAT_2_TEMP_HOTEND 240 
//Alexoko12 #define PREHEAT_2_TEMP_BED    110 
//Alexoko12 #define PREHEAT_2_FAN_SPEED     0 // Value from 0 to 255

//Alexoko12
#define PREHEAT_1_LABEL       "PLA"
#define PREHEAT_1_TEMP_HOTEND 190
#define PREHEAT_1_TEMP_BED     61
#define PREHEAT_1_FAN_SPEED     0 // Value from 0 to 255

#define PREHEAT_2_LABEL       "PLA Volcano"
#define PREHEAT_2_TEMP_HOTEND 240
#define PREHEAT_2_TEMP_BED     61
#define PREHEAT_2_FAN_SPEED     0 // Value from 0 to 255

#define PREHEAT_3_LABEL       "ABS"
#define PREHEAT_3_TEMP_HOTEND 210
#define PREHEAT_3_TEMP_BED    110
#define PREHEAT_3_FAN_SPEED     0 // Value from 0 to 255

#define PREHEAT_4_LABEL       "PETG"
#define PREHEAT_4_TEMP_HOTEND 233
#define PREHEAT_4_TEMP_BED    81
#define PREHEAT_4_FAN_SPEED     0 // Value from 0 to 255

#define PREHEAT_5_LABEL       "Leveling"
#define PREHEAT_5_TEMP_HOTEND 0
#define PREHEAT_5_TEMP_BED    40
#define PREHEAT_5_FAN_SPEED     0 // Value from 0 to 255

___________________________________________________________________

Cutout from "Marlin\Configuration_adv.h"
/**
 * User-defined menu items that execute custom GCode
 */
#define CUSTOM_USER_MENUS //Alexoko12
#if ENABLED(CUSTOM_USER_MENUS)
  #define CUSTOM_USER_MENU_TITLE "Preheat Tempretures" //Alexoko12 //#define CUSTOM_USER_MENU_TITLE "Custom Commands"
  #define USER_SCRIPT_DONE "M117 User Script Done"
  #define USER_SCRIPT_AUDIBLE_FEEDBACK
  //#define USER_SCRIPT_RETURN  // Return to status screen after a script

  #define USER_DESC_1 "Preheat for " PREHEAT_5_LABEL
  #define USER_GCODE_1 "M140 S" STRINGIFY(PREHEAT_5_TEMP_BED) "\nM104 S" STRINGIFY(PREHEAT_5_TEMP_HOTEND)

  #define USER_DESC_2 "Preheat for " PREHEAT_1_LABEL
  #define USER_GCODE_2 "M140 S" STRINGIFY(PREHEAT_1_TEMP_BED) "\nM104 S" STRINGIFY(PREHEAT_1_TEMP_HOTEND)

  #define USER_DESC_3 "Preheat for " PREHEAT_2_LABEL
  #define USER_GCODE_3 "M140 S" STRINGIFY(PREHEAT_2_TEMP_BED) "\nM104 S" STRINGIFY(PREHEAT_2_TEMP_HOTEND)

  #define USER_DESC_4 "Preheat for " PREHEAT_3_LABEL
  #define USER_GCODE_4 "M140 S" STRINGIFY(PREHEAT_3_TEMP_BED) "\nM104 S" STRINGIFY(PREHEAT_3_TEMP_HOTEND)

  #define USER_DESC_5 "Preheat for " PREHEAT_4_LABEL
  #define USER_GCODE_5 "M140 S" STRINGIFY(PREHEAT_4_TEMP_BED) "\nM104 S" STRINGIFY(PREHEAT_4_TEMP_HOTEND)

#endif

MKS-Sean, 2020-04-29:
@Alexoko12  
I think the problem of eeprom storage needs to be fixed by the marlin team. Of course, we will also test this problem, try to modify it, and then feedback to the marlin team.
Yes, we will add a leveling interface and a custom command interface, and some interfaces will gradually be added later.

mks-viva, 2020-12-11:
I will close this issues temporarily

closed 2020-12-11 by mks-viva