makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #19: MKS robin mini, marlin 2.0.5.3

Malderin, 2020-04-30:
Hello. I use marlin 2.0.5.3 with standard interface. Can you port this firmware to the mks robin mini board?

imberest, 2020-04-30:
Change board to MKS Robin MINI and it should work

Malderin, 2020-05-01:
> Change board to MKS Robin MINI and it should work

I tried, building error

MKS-Sean, 2020-05-05:
@Malderin This version does not yet support the TFT24 interface.

MKS-Sean, 2020-05-05:
@Malderin You can try to add the corresponding pin feet in pins_MKS_ROBIN_MINI.h.
Like this:
#define SPI_FLASH
#if ENABLED(SPI_FLASH)
	#define 	W25QXX_CS_PIN		PB12
	#define 	W25QXX_MOSI_PIN		PB15
	#define 	W25QXX_MISO_PIN		PB14
	#define 	W25QXX_SCK_PIN		PB13
#endif

Malderin, 2020-05-05:
Errors disappeared, but the firmware does not start, black screen

Malderin, 2020-05-05:
> @Malderin You can try to add the corresponding pin feet in pins_MKS_ROBIN_MINI.h.
> Like this:
> #define SPI_FLASH
> #if ENABLED(SPI_FLASH)
> #define W25QXX_CS_PIN	PB12
> #define W25QXX_MOSI_PIN	PB15
> #define W25QXX_MISO_PIN	PB14
> #define W25QXX_SCK_PIN	PB13
> #endif

errors disappeared, but the firmware does not start, black screen

MKS-Sean, 2020-05-06:
@Malderin Have you updated the mks_pic and mks_font files?

Malderin, 2020-05-06:
> @Malderin Have you updated the mks_pic and mks_font files?

Yes, I tried 2 options:
1. Stock the mks_pic and mks_font
2. Your the mks_pic and mks_font
After flashing your version, the screen does not turn on even normal marlin.

MKS-Sean, 2020-05-07:
@Malderin I will debug it when I have time.

Malderin, 2020-05-07:
> @Malderin I will debug it when I have time.
Thank! I will wait.

enlight3d, 2020-05-07:
Hello, same here, but after some minutes I get : OUTAGE RECOVERY PRINTER HALTED PLEASE RESTART

enlight3d, 2020-05-08:
Should also point out that I tried also with commenting  
`#define LCD_RESET_PIN    PC6`
as it was said in a comment inside pins_MKS_ROBIN_MINI.h but it didn't change anything except adding a white screen right after the boot. It stays white for half a second with that line commented and then go back to blank.

enlight3d, 2020-05-20:
@MKS-Sean any news ?

makerbase-mks, 2020-05-25:
@enlight3d You can try to comment out these two lines of code.
//#define POWER_LOSS_PIN                    PA2   // PW_DET
//#define PS_ON_PIN                         PA3   // PW_OFF

enlight3d, 2020-05-26:
@makerbase-mks just tested it. It does the same, white screen for half a second and then blank screen.

MKS-Sean, 2020-05-26:
@enlight3d Is your motherboard robin_mini + robin tft35?

enlight3d, 2020-05-26:
@MKS-Sean yes but with tft2.4

Malderin, 2020-07-25:
@MKS-Sean 
We were able to partially run this firmware. We transferred the display code from the MKS robin mini board. Then we scaled the screen size from 3.5 inches to 2.4.
The screen starts up, reacts to pressing, but when loading the firmware, icons from the folder mks_pic are not loaded . 
What could be the problem?

Malderin, 2020-07-25:
@MKS-Sean , 
Here's a video:
https://yadi.sk/i/TYI8KacRsVs2vw

mks-viva, 2020-12-11:
@Malderin 
Has this problem been resolved?

closed 2021-02-21 by MKS-Sean