makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #20: [Question] Setting Up TMC2208 on UART

VIXIVIXIV, 2020-05-05:
So, I've done some research, and I think I get the gist of adding UART to a Robin Nano setup. I'm using [this Instructables Guide](https://www.instructables.com/id/UART-This-Serial-Control-of-Stepper-Motors-With-th/), but it was written for Marlin V1 and not single-wire UART, and want to make sure I'm not missing anything important.

My plan, basically:

1. Uncomment the driver lines in Configuration.h (in my case, X, Y, Z, and E0) and set the driver type to TMC2208
2. Configure Configuration_adv.h more or less according to the guide
3. In the pins definition for the Robin Nano, add the necessary sections under #if ENABLED(HAVE_TMC2208) [specifically, #define X_SERIAL_TX_PIN, etc]
4. For the UART pins, use one of the redundant pins that UART makes unnecessary for each stepper  motor (DIAG, INDEX, MS1, MS2, and analog current setting VREF) in single pin UART mode (#define same pin for both SERIAL_TX_PIN and SERIAL_RX_PIN)
5. Install the TMC2208 and connect the PDN_UART pins to the pins specified in the header file

Anything that I'm missing? I want to make sure that reassigning, say, the DIAG pin won't cause any problems.

VIXIVIXIV, 2020-05-05:
Alright, done some more research:

1. I'm thinking of using the TMC2209 instead, but the wiring and code is probably basically the same.
2. Most of the pins made redundant by UART would not be connected to the microcontroller anyways. As such, which pins, if any, would be free to be connected for UART?

MKS-Sean, 2020-05-05:
@VIXIVIXIV 
You can configure your firmware like this：
[image]
You also need to define your UART pins in the pins_MKS_ROBIN_NANO.h file:
I also need to confirm whether these pins are correct.

// Software SPI pins for TMC2130 stepper drivers
//
#if ENABLED(TMC_USE_SW_SPI)
  #ifndef TMC_SW_MOSI
    #define TMC_SW_MOSI    PD14
  #endif
  #ifndef TMC_SW_MISO
    #define TMC_SW_MISO    PD1
  #endif
  #ifndef TMC_SW_SCK
    #define TMC_SW_SCK     PD0
  #endif
#endif

#if HAS_TMC_UART
  /**
   * TMC2208/TMC2209 stepper drivers
   *
   * Hardware serial communication ports.
   * If undefined software serial is used according to the pins below
   */
  //#define X_HARDWARE_SERIAL  Serial
  //#define X2_HARDWARE_SERIAL Serial1
  //#define Y_HARDWARE_SERIAL  Serial1
  //#define Y2_HARDWARE_SERIAL Serial1
  //#define Z_HARDWARE_SERIAL  Serial1
  //#define Z2_HARDWARE_SERIAL Serial1
  //#define E0_HARDWARE_SERIAL Serial1
  //#define E1_HARDWARE_SERIAL Serial1
  //#define E2_HARDWARE_SERIAL Serial1
  //#define E3_HARDWARE_SERIAL Serial1
  //#define E4_HARDWARE_SERIAL Serial1

  //
  // Software serial
  //

  #define X_SERIAL_TX_PIN                   PD5
  #define X_SERIAL_RX_PIN                   PD5

  #define Y_SERIAL_TX_PIN                   PD7
  #define Y_SERIAL_RX_PIN                   PD7

  #define Z_SERIAL_TX_PIN                  PD4
  #define Z_SERIAL_RX_PIN                  PD4

  #define E0_SERIAL_TX_PIN                 PD9
  #define E0_SERIAL_RX_PIN                 PD9

  #define E1_SERIAL_TX_PIN                 PD8
  #define E1_SERIAL_RX_PIN                 PD8

  // Reduce baud rate to improve software serial reliability
  #define TMC_BAUD_RATE 19200
#endif // TMC2208 || TMC2209

Miraculix200, 2020-05-05:
Some youtuber managed to connect all the stepper drivers to 2 pins (PA1 and PA6) to setup a serial connection. He used the jumpers on the RObin Nano to give each stepper a dedicated hardware address. Then it works. I'm not sure if it's UART or another type of serial connection however. And I don't understand polish, so I don't know what he did exactly. https://www.youtube.com/watch?reload=9&v=ZK9ttGcpiC4

VIXIVIXIV, 2020-05-05:
@Miraculix200 I had a chat with some folks in the Marlin Discord, and they said you can run 4 stepper drivers off one line in UART, just use the jumpers to set hardware addresses. Sound good, but ideally, I wouldn't have to use any of the pins from the 5th slot.

The instructable mentions the possibility of running the drivers in UART without the EN pin connected; if the EN pin isn't necessary, then I could try using that for UART instead. However, the instructable also mentions not being able to get it to work for some reason.

Miraculix200, 2020-05-06:
You can probably also use some of the pins of the wifi module slot instead of using the 5th stepper slot

swalsh11, 2020-06-14:
WIFI Pins method here apperently done here; https://www.reddit.com/r/3Dprinting/comments/gthxb6/mks_robin_nano_v12_diagram_for_5x_tmc2209_v30/

VIXIVIXIV, 2020-06-16:
Got the information I needed from the Marlin discord, I should be all set.

closed 2020-06-16 by VIXIVIXIV

faisalhusain007, 2020-06-16:
> Got the information I needed from the Marlin discord, I should be all set.
Care to share them, I am a bit stuck

swalsh11, 2020-06-16:
I am working through this too,  in the fork I am using,  I can edit, Pins_mks_robin_nano, to  #define SOFTWARE_SERIAL

this diabels E1, and re-uses the communications pins for UART.
this then defines: 
    #define X_SERIAL_TX_PIN                   PA3
    #define X_SERIAL_RX_PIN                   PA3
    #define Y_SERIAL_TX_PIN                   PA6
    #define Y_SERIAL_RX_PIN                   PA6
    #define Z_SERIAL_TX_PIN                   PA1
    #define Z_SERIAL_RX_PIN                   PA1
    #define E0_SERIAL_TX_PIN                  PE5
    #define E0_SERIAL_RX_PIN                  PE5

PA 3 is located in JC1_M2, EN2 Pin 
PA 6 is located in JC1_M2, STEP2 Pin
PA 1 is located in JC1_M2, DIR2 Pin 

I still have a bit of figuring out to do, Hope this helps.

faisalhusain007, 2020-06-17:
> I am working through this too, in the Marlin fork I am using, I can edit, Pins_mks_robin_nano, to #define SOFTWARE_SERIAL
> 
> this diables E1, and re-uses the communications pins for UART.
> this then defines:
> #define X_SERIAL_TX_PIN PA3
> #define X_SERIAL_RX_PIN PA3
> #define Y_SERIAL_TX_PIN PA6
> #define Y_SERIAL_RX_PIN PA6
> #define Z_SERIAL_TX_PIN PA1
> #define Z_SERIAL_RX_PIN PA1
> #define E0_SERIAL_TX_PIN PE5
> #define E0_SERIAL_RX_PIN PE5
> 
> PA 3 is located in JC1_M2, EN2 Pin
> PA 6 is located in JC1_M2, STEP2 Pin
> PA 1 is located in JC1_M2, DIR2 Pin
> (close to Xstop / E1)
> PE 5 is on the TCmax31855 header near THM header
> I still have a bit of figuring out to do, it is not compiling , but I Hope this helps.
> 
> Edit, Compiles in hardware_serial mode, using PA9.
> 
> I just joined Marlin Discord too, https://discord.gg/n5NJ59y

Thanks for the help, I was wondering if you have a resistor between tx/rx pins on your tmc?
Also when you say ' Compiles in hardware_serial mode, using PA9 '

This is in your config_adv right?  Can you share the tmc snippet of your comfig_adv too. Thank you.

swalsh11, 2020-06-17:
Reference here for Pins: and board scematic,.
https://github.com/makerbase-mks/MKS-Robin-Nano/tree/master/hardware/MKS%20Robin%20Nano%20V1.2_004

My Code:
https://github.com/swalsh11/Marlin-2.0.X-MKS-Robin-Nano/tree/Sapphire-Plus/Marlin

This compiles in both hardware serial and software serial mode,  but has ot been tested on my (son's) machine just yet

My Pins file: ( Where hardware or software serial mode is assigned)
https://github.com/swalsh11/Marlin-2.0.X-MKS-Robin-Nano/blob/Sapphire-Plus/Marlin/src/pins/stm32f1/pins_MKS_ROBIN_NANO.h

My References:
https://www.reddit.com/r/3Dprinting/comments/gthxb6/mks_robin_nano_v12_diagram_for_5x_tmc2209_v30/

https://www.gitmemory.com/issue/MarlinFirmware/Marlin/15873/552647130

https://www.youtube.com/watch?v=dOJbSrWVu_Q

faisalhusain007, 2020-06-17:
Thank you @swalsh11 really appreciate your help

khunpoum, 2020-12-04:
https://youtu.be/7ShcFKXrVHo

1nishikawa, 2021-02-20:
como configurar a placa rumba  tmc 2209 v3

1nishikawa, 2021-02-20:
[image]

1nishikawa, 2021-02-20:
[image]

1nishikawa, 2021-02-20:
[image]

1nishikawa, 2021-02-20:
[image]