makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #22: MKS Robin TFT

daniyo27, 2020-05-11:
Any chance for this to be ported over to MKS Robin TFT28/32 please?  There isn’t source code posted for the Robin TFT either!

Or does the existing code + original .bin icons work already for this board?

C0Bactan, 2020-05-11:
Hi I tried that and to convert bmp in bin files and could not get it to work.

EPDCenter, 2020-05-11:
Same problem here

fenelon, 2020-05-13:
I get a black blocky TFT after flashing (Robin Nano v.1.1). Is this only caused by images being the wrong dimensions or what? Anyway, some guidance would be much appreciated!

EPDCenter, 2020-05-13:
The Firmware 2.0 doesnt support for Robin TFT 32....BTT SKR really is much better, all source code is published...

MKS-Sean, 2020-05-13:
@daniyo27 The 28/32 screen display interface will increase in the future.
@C0Bactan Can you tell me your steps? And send me your bin file.
@fenelon You can copy the files in the Firmware folder to update.
@EPDCenter Our code is also all open source, and you are welcome to give some suggestions for development.

fenelon, 2020-05-13:
@MKS-Sean I have MKS Robin Nano v1.1 with 3.2 MKS TFT. Is there no support for it yet? I wonder if I can get Z_SAFE_HOMING to work, which is only supported in 2.0

EPDCenter, 2020-05-13:
> I get a black blocky TFT after flashing (Robin Nano v.1.1). Is this only caused by images being the wrong dimensions or what? Anyway, some guidance would be much appreciated!

Same problem with MKS Robin Nano 1.1 and TFT32 @MKS-Sean after 1 year for MKS Robin, Where is the documentation? Where is the Wiki explaining the parts of code?

artdezigner, 2020-05-13:
Yes, version 1.1 +TFT 32 is not normally supported: 
versions 1.0.0 - 1.0.4 do not have normal functionality + the radiator fan does not automatically turn on when the nozzle is heated. (My Tevo Nereus with MS Robin Nano V. 1.1 + Robin TFT 32 only works normally on version 1.0.1 from Tevo, this firmware is already a year old). 
Version 2.0 is only for TFT 35, but 2 months ago we promised under the video that there will be a firmware until the end of March. 
The Marlin + GUI version is also under TFT 35...
Marlin 2.0 bugfix with a standard interface for Robin Nano was adapted by simple facebook users. 

Very bad, MKS is a good company, but support for already released products is poor. Here is for example my question 16 days ago https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/issues/15 
Nothing has changed since then... WE NEED A FULL FUNCTION FIRMWARE WITH NORMAL GRAPHICAL INTERFACE, PLEASE MAKE IT!

EPDCenter, 2020-05-16:
Waiting a solution...

MKS-Sean, 2020-05-17:
@fenelon The 3.2-inch display interface will increase after the 3.5-inch firmware is stable

MKS-Sean, 2020-05-17:
@EPDCenter The marlin 2.0 firmware already supports MKS Robin

MKS-Sean, 2020-05-17:
@artdezigner Our idea is to wait for the 3.5-inch firmware to stabilize before adding a 3.2-inch interface.

EPDCenter, 2020-05-17:
@MKS-Sean you can publish other branch with the gui for 3.2 and we can help you with the developing

makerbase-mks, 2020-05-18:
@EPDCenter Of course, I have created the Robin-TFT32 branch.

EPDCenter, 2020-05-18:
@MKS-Sean thanks ;), can you upload the "Alpha" Code of 3.2 TFT please? Because It is the same code of master branch.

EPDCenter, 2020-05-19:
@makerbase-mks @MKS-Sean any answer?

MKS-Sean, 2020-05-20:
@EPDCenter Do you mean I uploaded a 3.2-inch basic version?

daniyo27, 2020-05-20:
@MKS-Sean yes, can you please upload a basic version for TFT32? The other branch is identical to the TFT35 branch.

MKS-Sean, 2020-05-20:
@daniyo27 @EPDCenter ok, I will upload a code as soon as possible.

daniyo27, 2020-05-20:
@MKS-Sean thank you! We would love to help you ASAP too.

daniyo27, 2020-05-28:
@MKS-Sean any news? Love to help asap.

EPDCenter, 2020-06-01:
@MKS-Sean @makerbase-mks any news?

MKS-Sean, 2020-06-05:
@EPDCenter I heard some information, some developers support 3.2-inch color display based on marlin, can this meet your needs?
https://github.com/MarlinFirmware/Marlin/issues/18129

EPDCenter, 2020-06-13:
@makerbase-mks @MKS-Sean 1 month waiting..the update

MKS-Sean, 2020-06-14:
@EPDCenter I hope to do this in June.

aer-neo, 2020-07-04:
@makerbase-mks @MKS-Sean hi, is there any tentative date for MKS Robin TFT28 / 32 support? Thank you.

MKS-Sean, 2020-07-04:
@aer-neo We are currently seeking help to do this.

artdezigner, 2020-07-20:
> @daniyo27 The 28/32 screen display interface will increase in the future. (that was 2 months ago)
...
...
...
So what? cheated again? I think that we will not get support for display 32. We need to switch to the boards of those manufacturers who really support their product and care about their customers.

aer-neo, 2020-07-25:
> @daniyo27 The 28/32 screen display interface will increase in the future. (that was 2 months ago)
> ...
> ...
> ...
> So what? cheated again? I think that we will not get support for display 32. We need to [...]

I hope they'll be able to say they're not going to support TFT32 so they don't waste time waiting.

eghbalmohsen, 2020-11-03:
I am totally disappointed since 3.2" TFT_V2.0 is not supported. It is better to be honest customers before buying your products, for example, All our effort is for tft35, we don't have any plan for smaller displays....

This is not a good policy, I have bought your products for more than a year and I can't get them to work with my desired firmware...

artdezigner, 2020-11-20:
the manufacturer deceived us, there is no firmware

SHBnik, 2020-11-25:
Hi,
I think I found the problem. I have MKS Robin nano v1.2 and these LCDs : 2.4 v1 , 3.2 v2 and 3.5 v2.
The robin nano board just update it firmware when it is connected to the 3.5 inch LCD, it cannot accept new firmware if it is not connected to any LCD or 2.4 or 3.2 LCD. but when i connect the 3.5 LCD and configure mt marlin for the 3.2 or 2.4, robin nano show the uploading page and after finishing that shows smaller size LCD in 3.5, then when you connect the other LCDs it works fine with out bug. 
I think in the boot loader code the have condition the check the connection between the lcd and the board to show that uploading page and if it is not available it will stuck in there. 
the 3.5 driver is ili9488 but the 3.2 and 2.4 have a st7789  driver. so the communication drivers are different and cause the boat loader get stuck in a line.
I am trying to simulate the communication of an ili9488 with an arduino uno to force the robin nano board to pass this step and upload the code for other lcds when there is no 3.5 lcd.
The better way is, the MKS give us the boot loader so that we can fix that problem and upload the new boot loader with the jtag connector on the board.  
The other mks board like mks robin that supports this kind of lcds do not have this problem in the boot loader.