makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #24: [BUG]  queue' was not declared in this scope

le3tspeak, 2020-05-16:
819/5000
Hi everyone
Once you get the function: #define AUTO_BED_LEVELING_BILINEAR
The following error is activated:

Marlin \ src \ lcd \ extui \ lib \ mks_ui \ src \ draw_tool.cpp: In function 'void event_handler (lv_obj_t *, lv_event_t)':
Marlin \ src \ lcd \ extui \ lib \ mks_ui \ src \ draw_tool.cpp: 77: 4: error: 'queue' was not declared in this scope
     queue.enqueue_one_P (PSTR ("G28"));
     ^ ~~~~
*** [.pio \ build \ mks_robin_nano35 \ src \ src \ lcd \ extui \ lib \ mks_ui \ src \ draw_tool.cpp.o] Error 1

What occurs in the draw_tool.ccp in line 75-83

  else if (event == LV_EVENT_RELEASED) {
#if ENABLED (AUTO_BED_LEVELING_BILINEAR)
queue.enqueue_one_P (PSTR ("G28"));
queue.enqueue_one_P (PSTR ("G29"));
#else
uiCfg.leveling_first_time = 1;
lv_clear_tool ();
lv_draw_manualLevel ();
#endif

solutions already available?

greetings

darksiah, 2020-05-17:
I had the same problem and I solved it by adding the following line in the drawtool.cpp file:

#include "../../../../../gcode/queue.h"

MKS-Sean, 2020-05-17:
@le3tspeak This should not introduce the queue.h header file, I will add it.

mks-viva, 2020-11-19:
I will close this issues temporarily.

closed 2020-11-19 by mks-viva