makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #27: ADVANCED_PAUSE_FEATURE currently requires an LCD controller or EMERGENCY_PARSER

chenxiaowei107, 2020-05-27:
编译的时候提示错误，应该是lcd菜单没有定义这2个功能吧。求增加。。
[image]

MKS-Sean, 2020-05-29:
My later version will not use this option when using the color UI, because there will be some conflicts with the original content of the marlin firmware.

MKS-Sean, 2020-05-29:
I will increase the function of setting the pause position online, and the refueling function

chenxiaowei107, 2020-05-29:
> I will increase the function of setting the pause position online, and the refueling function

Also The bed levling is not in the lcd menu.

MKS-Sean, 2020-05-29:
@chenxiaowei107 Leveling interface already exists

swalsh11, 2020-06-14:
Same bug here on compile

yousry, 2020-07-24:
Same problem here with V1.1 of the board:

In **Configuration.h**
```
#define FIL_RUNOUT_PIN   PA4
#define FILAMENT_RUNOUT_SENSOR
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
...
#define NOZZLE_PARK_FEATURE
...
// Necessary for sanity check 
#define EXTENSIBLE_UI

//If TFT_LITTLE_VGL_UI is enabled, there is no need to turn on FSMC_GRAPHICAL_TFT.
#define TFT_LITTLE_VGL_UI
//Robin_nano_v2.0 motherboard touch screen is spi interface, you need to enable this item.
#define SPI_GRAPHICAL_TFT	
```
in **Configuration_adv.h**

```
define ADVANCED_PAUSE_FEATURE
```

I ended up with this error:
```
Marlin/src/lcd/extui/lib/mks_ui/src/draw_dialog.cpp: In function 'void btn_ok_event_cb(lv_obj_t*, lv_event_t)':
Marlin/src/lcd/extui/lib/mks_ui/src/draw_dialog.cpp:123:3: error: 'pause_menu_response' was not declared in this scope
```

closed 2022-05-16 by solawc