makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #28: OctoPrint print job not showing on the printers screen

ttutuncu, 2020-05-28:
Hi,

I own a Two Trees Bluer 3D Printer.
Mainboard: MKS Robin Nano 1.2
Board Firmware: MKS Robin Nano Firmware version 2.0.2
OctoPi Version: 0.17.0, running on Raspberry Pi 4 Model B Rev 1.1.
OctoPrint version: 1.4.0

The problem I have is when I start a print job from OctoPrint my printer starts printing it but the printers screen doesn't show anything but the standard menu (as if no printing has started). I want it to show the printing menu (operations and status) like when I start a print from SD Card. Is there anyway I can trigger the printing status on my firmware to show its normal printing screen?

Thanks.

MKS-Sean, 2020-05-29:
This function needs to be modified to be supported by firmware, what do you think?
https://github.com/makerbase-mks/MKS-Robin-Nano-Firmware

ttutuncu, 2020-05-29:
Hi @MKS-Sean,

I have asked the OctoPrint Community as well:
https://community.octoprint.org/t/printer-doesnt-show-printing-status/20393

Maybe M117 and M73 gcode can be used to trigger the printing interface? OctoPrint has plugins for adding these commands to gcode.

UnleashSpirit, 2020-12-07:
Same issue here
It's look like octpi does not "taking control" of board so can't show any print information and neither can't acces to print function (like babystepping)

madurani, 2021-03-24:
Is there some progress?

just-trey, 2021-09-14:
I would  like this feature as well