makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #31: Image shift in "more interface"(ReadyPrint>Tool>More) screen in Robin TFT35

vispell, 2020-06-04:
I followed the manual instructions and named the file bmp_custom1.bin - bmp_custom7.bin.
However, when I install it, the images of bmp_custom1.bin appear in the first and second two places, and the numbers from the third to the seventh are out of alignment.
** Please see the attached image.
Therefore, bmp_custom7.bin will not be displayed.
The command specified in cfg is correct.
[image]

MKS-Sean, 2020-06-05:
@vispell What is the firmware you use?

MKS-Sean, 2020-06-06:
@vispell I fixed this problem, you can update the firmware to test it
https://github.com/makerbase-mks/MKS-Robin-Nano/tree/master/Nano%20firmware%20v2.x.x

vispell, 2020-06-06:
Thank you for your update. I will look up the corrections and study them. Thank you so much.

closed 2020-06-06 by vispell