makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #33: [question] How to compile

ihouses, 2020-06-11:
Hi am trying to compile the code with Visual Studio and platform IO. 
I select the mks_robin_nano35->build and when I compile I get some errors like:

.pio\libdeps\mks_robin_nano35\Adafruit BusIO_ID6214\Adafruit_I2CDevice.cpp: In member function 'bool Adafruit_I2CDevice::read(uint8_t*, size_t, bool)':
.pio\libdeps\mks_robin_nano35\Adafruit BusIO_ID6214\Adafruit_I2CDevice.cpp:160:79: error: no matching function for call to 'TwoWire::requestFrom(uint8_t, uint8_t, uint8_t)'

What I am doing wrong?

MKS-Sean, 2020-06-14:
[platformio]
src_dir      = Marlin
boards_dir   = buildroot/share/PlatformIO/boards
default_envs = mks_robin_nano35

[common]
default_src_filter = +<src/*> -<src/config> -<src/HAL> +<src/HAL/shared>
extra_scripts = pre:buildroot/share/PlatformIO/scripts/common-cxxflags.py
build_flags = -fmax-errors=5 -g -D__MARLIN_FIRMWARE__ -fmerge-all-constants
lib_deps =
  LiquidCrystal
  TMCStepper@>=0.6.2,<1.0.0
  Adafruit NeoPixel
  U8glib-HAL=https://github.com/MarlinFirmware/U8glib-HAL/archive/bugfix.zip
  #Adafruit_MAX31865=https://github.com/adafruit/Adafruit_MAX31865/archive/master.zip
  LiquidTWI2=https://github.com/lincomatic/LiquidTWI2/archive/master.zip
  Arduino-L6470=https://github.com/ameyer/Arduino-L6470/archive/0.8.0.zip
  SailfishLCD=https://github.com/mikeshub/SailfishLCD/archive/master.zip
  SailfishRGB_LED=https://github.com/mikeshub/SailfishRGB_LED/archive/master.zip
  SlowSoftI2CMaster=https://github.com/mikeshub/SlowSoftI2CMaster/archive/master.zip

MKS-Sean, 2020-06-14:
You can block the Adafruit_MAX31865 library and delete the .pio and .vscode folders and try to compile again.

swalsh11, 2020-06-14:
I have the same bug,

themacboy, 2020-06-17:
You can also downgrade to library verrsion 1.1.0:

 Adafruit_MAX31865=https://github.com/adafruit/Adafruit_MAX31865/archive/1.1.0.zip

swalsh11, 2020-06-18:
Thanks, that Worked,  
in Bugfix, they have, that works too, and probably shoud be merged into this

lib_deps =
  LiquidCrystal
  TMCStepper@>=0.6.2
  Adafruit NeoPixel
  Adafruit MAX31865 library@>=1.1,<1.2
  U8glib-HAL=https://github.com/MarlinFirmware/U8glib-HAL/archive/bugfix.zip
  LiquidTWI2=https://github.com/lincomatic/LiquidTWI2/archive/master.zip
  Arduino-L6470=https://github.com/ameyer/Arduino-L6470/archive/0.8.0.zip
  SailfishLCD=https://github.com/mikeshub/SailfishLCD/archive/master.zip
  SailfishRGB_LED=https://github.com/mikeshub/SailfishRGB_LED/archive/master.zip
  SlowSoftI2CMaster=https://github.com/mikeshub/SlowSoftI2CMaster/archive/master.zip

swalsh11, 2020-06-18:
Putting in my First Pull request to try and get this fixed.

themacboy, 2020-06-23:
Im getting a general compilation failure when trying to compile the TFT35.

Not sure how I must configure it. I tryed to use your defaults or the original Marlin last bugfixes version.

But never succes.

Can someone teach me? Thbaks in advance !

themacboy, 2020-06-23:
Other question ...

 all modifcations posted in the original Marlin repository are automatically imported to this repository? if yes, what branch?

makerbase-mks, 2020-06-24:
@themacboy You can block the Adafruit_MAX31865 library and delete the .pio and .vscode folders and try to compile again.
The original marlin will not be automatically imported into this repository , we will keep the same with the original marlin later.

mks-viva, 2020-12-11:
I will close this issues temporarily

closed 2020-12-11 by mks-viva