makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #34: Serial communication over USB with a Flsun Q5

mgcrea, 2020-06-15:
I'm trying to send some GCode over USG to calibrate the printer and it does not work.

I can properly connect at 115200bits, but when I try to send a command like `M503`, the board does spam:

```
Resend:1
ok

Resend:1
ok

Resend:1
ok

Resend:1
ok

Resend:1
ok

...
```

Also tried `Repetier-Host` on macOS but it does not seem to work.

Any idea on how to properly communicate with the printer?

makerbase-mks, 2020-06-16:
Hi! M500-M503 is effective when you enabled eeprom, you can try to it

Whiskaya, 2020-09-20:
> Hi! M500-M503 is effective when you enabled eeprom, you can try to it

How can you "enable EEPROM"? Is there an EEPROM menu in the Q5? (Like in a standard Marlin printer por example?)

I thought M503 was enabled/disabled when compiling Marlin, and "unknown command" shows up when that command hasn't been compiled.

mks-viva, 2020-12-11:
@Whiskaya 
In Configuration.h file: enable #define EEPROM_SETTINGS
[image]

closed 2021-02-21 by MKS-Sean