makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #35: Add wifi support for normal marlin interface

rosellonsito, 2020-06-15:
How can I use the normal marlin with the wifi module?

makerbase-mks, 2020-06-16:
Currently, MKS Robin WIFI is temporarily incompatible, we will continue to pay attention to this issue

dgorbunov, 2020-07-06:
+1 on this. I much prefer Marlin over the stock firmware and one of the reasons I bought this board was for WiFi support

Loreagle, 2020-07-07:
+1! I think that with the new firmware with MKS interface they have one reason more to add the MKS wifi support! That would be very nice for all of us that bought wifi module for this board :)

nictronik99, 2020-07-17:
+1I think that selling a board with wifi socket that cannot be used is almost a scam.the best option would be to integrate it in Marlin but in the meantime you should at least make an MKS firmware like the one for Robin nano v1.2 also on v2 !!

rosellonsito, 2020-07-24:
> Currently, MKS Robin WIFI is temporarily incompatible, we will continue to pay attention to this issue

What you can do is like the MksWifi.bin, provide one to flash the esp3d. It would be AWESOME!

mks-viva, 2020-12-11:
Haha! MKS Robin Nano has supported MKS Robin Wifi when use Marlin, refer to: 
Firmware: https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware
MKS Robin Nano V2.x: https://github.com/makerbase-mks/MKS-Robin-Nano-V2.X/wiki/Use-MKS_Robin_WIFI-on-Robin-Nano
MKS Robin Nano V1.x: https://github.com/makerbase-mks/MKS-Robin-Nano-V1.X/wiki/MKS_Robin_WIFI

closed 2020-12-11 by mks-viva