makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #36: Compatibility with an Flsun Q5 with basic touch screen

mgcrea, 2020-06-15:
Would it be possible to flash it on the MKS Robin Nano v1.2 coming inside an Flsun Q5? Looks like it has a dumb (no IC) TFT24 attached.

makerbase-mks, 2020-06-16:
This may not work, because the pin is different

Whiskaya, 2020-09-20:
> This may not work, because the pin is different

Does that mean that "it WON'T work" or that "it MAY work"?

If it "MAY" work, what do I have to do to compile the right version for a stock Q5?
I've tried several versions so far and some work, but invert the screen, and others just scramble the whole TFT.

If it "WON'T", does that mean that the only way to use run Marlin is to change the board and screen altogether?

Quas7, 2020-09-25:
works fine for me with latest Marlin bugfix-2.0.x.
#define TFT_320x240
#define TFT_DRIVER AUTO

Whiskaya, 2020-09-29:
> works fine for me with latest Marlin bugfix-2.0.x.
> #define TFT_320x240
> #define TFT_DRIVER AUTO

Ok so I finally got it working with the latest bugfix, BUT... the touchscreen functionality is inverted.
Tried #GRAPHICAL_TFT_ROTATE_180 and adjusting the touchscreen calibration values, but nothing happens.

https://github.com/MarlinFirmware/Marlin/issues/19544

Any ideas?

Quas7, 2020-09-29:
in /src/inc/Conditional_LCS.h it is stated: 

// FSMC/SPI TFT Panels
#if EITHER(FSMC_GRAPHICAL_TFT, SPI_GRAPHICAL_TFT)
  #define TFT_SCALED_DOGLCD 1
#endif

Maybe, you just remove the #if TFT_SCALED_DOGLCD the rotation is enabled (or you get a bunch of compile errors...).

There is also a pull request for an older Marlin-fork fixing this maybe: https://github.com/AndersSahlman/Marlin/pull/2

danielr1996, 2020-10-05:
> 
> 
> works fine for me with latest Marlin bugfix-2.0.x.
> #define TFT_320x240
> #define TFT_DRIVER AUTO

Can you share the configuration you used? I think you're referring to the bugfix branch from the official Marlin repo, however the configuration file from this repository doesn't compile with the error `#error "You are using an old Configuration.h file, update it before building Marlin."`

Quas7, 2020-10-05:
@danielr1996 
Sure, here it is. I think, the config works for bugfix 2.0.6.1 a52afd2ae

[Marlin_2.0.x-bugfx_Q5_TMC220x_Standalone.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/5328683/Marlin_2.0.x-bugfx_Q5_TMC220x_Standalone.zip)

I also have a more recent config for bugfix e8177735a  that includes TMC-UART, dual y-split extruder setup and what else I can not remember but did not touch the display setting, I think. 
But maybe, it is easier to rollback these modifications to your setup than using the outdated one above?
[Config_bugfix-0e8177735a.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/5328719/Config_bugfix-0e8177735a.zip)

danielr1996, 2020-10-05:
> @danielr1996
> Sure, here it is. I think, the config works for bugfix 2.0.6.1 a52afd2ae [...]

Thanks for you're help. I finally got it working. The key was to comment the following lines:
```//#define XPT2046_X_CALIBRATION 12316
  //#define XPT2046_Y_CALIBRATION -8981
  //#define XPT2046_X_OFFSET        -43
  //#define XPT2046_Y_OFFSET        257
```

Otherwise the offsets were scrambled and presses on the thouchscreen were noticed roughly 2cm lower than where I pressed.

eduwinner-1, 2020-11-14:
Hi, I', trying to install Marlin 2.0 in the Flsun Q5 but I am not able to fix the screen resolution neither rotate it.

As you know, Flsun Q5 has got the Robin Nano v1.2 board.

Is there any guide or configuration for downloading that I can try?

Thanks!

Quas7, 2020-11-17:
you tried to look into config settings from above already?
It looks like the Q5 has different TFTs depending on production batches so a look and photo from the backside of your TFT is likely needed.

eduwinner-1, 2020-11-17:
Hi @Quas7, I was able to install last Marlin 2.0 but I don't like very much the interface. I also tried with your Configuration.h and Marlin 2.0.6.1, it was successful but same interface.

I am now trying to install MKS Robin Nano v1.x interface with Marlin 2.0 but I can't fix the screen resolution neither orientation

[image]

Quas7, 2020-11-18:
@eduwinner-1 the icons are in seperate folders and need to be uploaded to the SD as well - google for marlin tft icons and grafics folder. Orientation and screen offset needs some tweaking depending on the actual TFT model.