makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #39: Add msc USB composite device

faisalhusain007, 2020-06-18:
Would like to see msc USB composite device added, like on the BTT stm32f1;
Should we follow the BTT implementation?

faisalhusain007, 2020-06-19:
Here is the main resource I am referring to
https://github.com/MarlinFirmware/Marlin/pull/15180

makerbase-mks, 2020-06-27:
@faisalhusain007 Is the USB composite device convenient to change the files of the sd card?

faisalhusain007, 2020-06-27:
yes especially for flashing then issue M997, I have connected the board to my pi but i can not see it using sudo blkid -o list  on pins I have included the following 
// SD Card
//
#ifndef SDCARD_CONNECTION
  #define SDCARD_CONNECTION              ONBOARD
#endif

#define SDIO_SUPPORT
#define SD_DETECT_PIN                       PD12

//

and added the relevent sections to config.h and config_adv.h too.
When compiling I include the -usb composite .... in the platform.ini too

I have this working on BTT SKR 1.3 and E3mini but cant get it to work on my MKS robin board.

makerbase-mks, 2020-06-27:
The robin nano motherboard needs to change the hardware to support this function. It requires a specific pin to connect to the USB interface, but it is not yet.

faisalhusain007, 2020-06-27:
could you provide more details, perhaps if it is only one pin, maybe we can use a wire to connect it, i take it your referring to a pin on the processor ?

chrisdaloa, 2020-10-30:
Any news? I'm interested in. Thanks.

Malderin, 2020-10-30:
> Any news? I'm interested in. Thanks.

https://yadi.sk/i/Ogz5Q_HT-PcpBA

MKS-Sean, 2021-02-21:
MKS Robin nano v3 already supports

closed 2021-02-21 by MKS-Sean