makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #40: how to upload this firmware to the board?

honey2pie, 2020-06-21:


makerbase-mks, 2020-06-22:
1. Copy Robin_nano35.bin to TF card
2. TF Card Insert to Nano board
3. Repower on

themacboy, 2020-06-23:
The first step is compile the firmware and copy the result .bin file to TF card and restart.

But compile is not always easy as spected ! ;)

honey2pie, 2020-06-24:
I copied Robin_nano35.bin including the 2 folder, mks font and mks pic, to SD card.
the upgrading process seem to working good. I can home the printer, comunicate to the printer with my PC with M command, but I cannot print file from SD card. when I excute the print command form the touch screen, the print pause itself and cannot resume to print. 
Is there any change should I change in the Configuration file to make the printer work?

makerbase-mks, 2020-06-27:
@honey2pie 
What configuration did you modify? If you can provide me with media information, I will better solve this problem.
@themacboy 
Which error was reported when you compiled, you can take a screenshot to show me. I don’t know if it is similar to this problem.
[#33 ](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/issues/33)

mks-viva, 2020-12-11:
I will close this issues temporarily

closed 2020-12-11 by mks-viva