makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #42: Compilation error

user32rtu, 2020-06-28:
**I have been getting this error for a long time. She still appears. Compiling for the default configuration.**

C:\Users\Sasha\.platformio\lib\Adafruit BusIO_ID6214\Adafruit_I2CDevice.cpp: In member function 'bool Adafruit_I2CDevice::read(uint8_t*, size_t, bool)':
C:\Users\Sasha\.platformio\lib\Adafruit BusIO_ID6214\Adafruit_I2CDevice.cpp:160:79: error: no matching function for call to 'TwoWire::requestFrom(uint8_t, uint8_t, uint8_t)'
   size_t recv = _wire->requestFrom((uint8_t)_addr, (uint8_t)len, (uint8_t)stop);
                                                                               ^
In file included from C:\Users\Sasha\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/Wire.h:42:0,
                 from C:\Users\Sasha\.platformio\lib\Adafruit BusIO_ID6214/Adafruit_I2CDevice.h:1,
                 from C:\Users\Sasha\.platformio\lib\Adafruit BusIO_ID6214\Adafruit_I2CDevice.cpp:1:
C:\Users\Sasha\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/utility/WireBase.h:101:11: note: candidate: uint8 WireBase::requestFrom(uint8, int)
     uint8 requestFrom(uint8, int);
           ^~~~~~~~~~~
C:\Users\Sasha\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/utility/WireBase.h:101:11: note:   candidate expects 2 arguments, 3 provided
C:\Users\Sasha\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/utility/WireBase.h:106:11: note: candidate: uint8 WireBase::requestFrom(int, int)
     uint8 requestFrom(int, int);
           ^~~~~~~~~~~
C:\Users\Sasha\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/utility/WireBase.h:106:11: note:   candidate expects 2 arguments, 3 provided
*** [.pio\build\mks_robin_nano35\lib923\Adafruit BusIO_ID6214\Adafruit_I2CDevice.cpp.o] Error 1
C:\Users\Sasha\.platformio\lib\Adafruit BusIO_ID6214\Adafruit_SPIDevice.cpp: In member function 'void Adafruit_SPIDevice::transfer(uint8_t*, size_t)':
C:\Users\Sasha\.platformio\lib\Adafruit BusIO_ID6214\Adafruit_SPIDevice.cpp:99:31: error: no matching function for call to 'SPIClass::transfer(uint8_t*&, size_t&)'
     _spi->transfer(buffer, len);
                               ^
In file included from C:\Users\Sasha\.platformio\lib\Adafruit BusIO_ID6214/Adafruit_SPIDevice.h:1:0,
                 from C:\Users\Sasha\.platformio\lib\Adafruit BusIO_ID6214\Adafruit_SPIDevice.cpp:1:
Marlin\src\HAL\STM32F1/SPI.h:268:11: note: candidate: uint8_t SPIClass::transfer(uint8_t) const
   uint8_t transfer(uint8_t data) const;
           ^~~~~~~~
Marlin\src\HAL\STM32F1/SPI.h:268:11: note:   candidate expects 1 argument, 2 provided
*** [.pio\build\mks_robin_nano35\lib923\Adafruit BusIO_ID6214\Adafruit_SPIDevice.cpp.o] Error 1
Indexing .pio\build\mks_robin_nano35\libc3e\libUSBComposite.a
In file included from C:\Users\Sasha\.platformio\lib\Adafruit MAX31865 library_ID1253\Adafruit_MAX31865.cpp:17:0:
C:\Users\Sasha\.platformio\lib\Adafruit MAX31865 library_ID1253\Adafruit_MAX31865.h:84:32: warning: passing NULL to non-pointer argument 1 of 'Adafruit_SPIDevice::Adafruit_SPIDevice(int8_t, uint32_t, BitOrder, uint8_t, SPIClass*)' [-Wconversion-null]
   Adafruit_SPIDevice spi_dev = NULL;
                                ^~~~
============================================================================================= [FAILED] Took 205.70 seconds =============================================================================================

user32rtu, 2020-06-29:
Configure someday platformio.ini with working versions of libraries.

nickrodie, 2020-07-02:
You can try comment out the adafruit line in platformio or amend it as per https://reprap.org/forum/read.php?415,874307 See Issue 2

mks-viva, 2020-12-11:
Has this problem been resolved?

closed 2021-02-21 by MKS-Sean