makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #43: Missing font

tubededentifrice, 2020-06-30:
Thanks for fixing the compilation, at least.
Now when compiling it seems to be working, except there is no font displayed (ie. everything written on the display is missing).
So I can see all the icons, but not the text underneath, which is quite problematic especially for the settings where everything is empty...

I suspect it might be because I don't have the "chinese" (as the font  font installed on my computer, which font should be embedded into the repo, though I haven't found which font exactly (wenquanyi_12pt?) to put where.

TimCarrDP, 2020-07-02:
Same issue. I managed to get it working once, but I can't remember what I did. Possibly including the font and images folder when flashing?

makerbase-mks, 2020-07-03:
@tubededentifrice @TimCarrDP 
https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/tree/master/Firmware
You need to update and put the mks_font folder on the sd card, then connect to the motherboard, and the font file will be updated after power on.

tubededentifrice, 2020-07-03:
Indeed, it works when adding those two folders, thanks!

closed 2020-07-03 by tubededentifrice