makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #44: error while compiling

nictronik99, 2020-06-30:
hello i have a few errors while compiling on platformIO, 
the file I opened with the "open project" button is the pure one, with no modifications that I downloaded from here. the errors are the following:

In file included from Marlin\src\HAL\STM32F1\dogm\../../../inc/../HAL/./STM32F1/HAL.h:36,
                 from Marlin\src\HAL\STM32F1\dogm\../../../inc/../HAL/HAL.h:26,
                 from Marlin\src\HAL\STM32F1\dogm\../../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\STM32F1\dogm\u8g_com_stm32duino_fsmc.cpp:29:
Marlin\src\HAL\STM32F1\dogm\../../../inc/../HAL/./STM32F1/fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\mks_robin_nano35\src\src\HAL\STM32F1\dogm\u8g_com_stm32duino_fsmc.cpp.o] Error 1
In file included from Marlin\src\HAL\STM32F1\../../inc/../HAL/./STM32F1/HAL.h:36,
                 from Marlin\src\HAL\STM32F1\../../inc/../HAL/HAL.h:26,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\STM32F1\onboard_sd.cpp:15:
Marlin\src\HAL\STM32F1\../../inc/../HAL/./STM32F1/fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\mks_robin_nano35\src\src\HAL\STM32F1\onboard_sd.cpp.o] Error 1
In file included from Marlin\src\HAL\shared\../../inc/../HAL/./STM32F1/HAL.h:36,
                 from Marlin\src\HAL\shared\../../inc/../HAL/HAL.h:26,
                 from Marlin\src\HAL\shared\../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\shared\HAL_spi_L6470.cpp:28:
Marlin\src\HAL\shared\../../inc/../HAL/./STM32F1/fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\mks_robin_nano35\src\src\HAL\shared\HAL_spi_L6470.cpp.o] Error 1
In file included from Marlin\src\HAL\shared\backtrace\../../../core/../inc/../HAL/./STM32F1/HAL.h:36,
                 from Marlin\src\HAL\shared\backtrace\../../../core/../inc/../HAL/HAL.h:26,
                 from Marlin\src\HAL\shared\backtrace\../../../core/../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\shared\backtrace\../../../core/serial.h:24,
                 from Marlin\src\HAL\shared\backtrace\backtrace.cpp:29:
Marlin\src\HAL\shared\backtrace\../../../core/../inc/../HAL/./STM32F1/fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\mks_robin_nano35\src\src\HAL\shared\backtrace\backtrace.cpp.o] Error 1
In file included from Marlin\src\HAL\shared\backtrace\../../../inc/../HAL/./STM32F1/HAL.h:36,
                 from Marlin\src\HAL\shared\backtrace\../../../inc/../HAL/HAL.h:26,
                 from Marlin\src\HAL\shared\backtrace\../../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\shared\backtrace\unwmemaccess.cpp:18:
Marlin\src\HAL\shared\backtrace\../../../inc/../HAL/./STM32F1/fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory     
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\mks_robin_nano35\src\src\HAL\shared\backtrace\unwmemaccess.cpp.o] Error 1
====================================================== [FAILED] Took 47.98 seconds ======================================================

Environment               Status    Duration
------------------------  --------  ------------
mega2560                  IGNORED
mega1280                  IGNORED
rambo                     IGNORED
FYSETC_F6_13              IGNORED
FYSETC_F6_14              IGNORED
sanguino644p              IGNORED
sanguino1284p             IGNORED
melzi                     IGNORED
melzi_optiboot            IGNORED
at90usb1286_cdc           IGNORED
at90usb1286_dfu           IGNORED
DUE                       IGNORED
DUE_USB                   IGNORED
DUE_debug                 IGNORED
LPC1768                   IGNORED
LPC1769                   IGNORED
STM32F103RC               IGNORED
STM32F103RC_fysetc        IGNORED
STM32F103RC_btt           IGNORED
STM32F103RC_btt_USB       IGNORED
STM32F103RC_btt_512K      IGNORED
STM32F103RC_btt_512K_USB  IGNORED
STM32F103RE               IGNORED
STM32F103RE_btt           IGNORED
STM32F103RE_btt_USB       IGNORED
STM32F4                   IGNORED
STM32F7                   IGNORED
ARMED                     IGNORED
STM32F103VE_GTM32         IGNORED
STM32F103VE_longer        IGNORED
mks_robin_mini            IGNORED
mks_robin_nano35          FAILED    00:00:47.976
mks_robin                 IGNORED
mks_robin_pro35           IGNORED
mks_robin_lite            IGNORED
mks_robin_lite3           IGNORED
jgaurora_a5s_a1           IGNORED
STM32F103CB_malyan        IGNORED
chitu_f103                IGNORED
STM32F401VE_STEVAL        IGNORED
FLYF407ZG                 IGNORED
FYSETC_S6                 IGNORED
STM32F407VE_black         IGNORED
BIGTREE_SKR_PRO           IGNORED
BIGTREE_GTR_V1_0          IGNORED
BIGTREE_BTT002            IGNORED
teensy31                  IGNORED
teensy35                  IGNORED
esp32                     IGNORED
linux_native              IGNORED
SAMD51_grandcentral_m4    IGNORED
rumba32_f446ve            IGNORED
rumba32_mks               IGNORED
include_tree              IGNORED
================================================= 1 failed, 0 succeeded in 00:00:47.976 =================================================
The terminal process terminated with exit code: 1

makerbase-mks, 2020-07-03:
@nictronik99 Did you try to update to the latest code and compile it?

nictronik99, 2020-07-03:
> @nictronik99 Did you try to update to the latest code and compile it?

yes but nothing change.

nictronik99, 2020-07-03:
@makerbase-mks https://github.com/MarlinFirmware/Marlin/issues/18505?fbclid=IwAR2C1ntx9bGu3D1PlJnOcbRrnT2XohEe6Fy6FluXdsrlZoxSaMRv53Rbq6A#issuecomment-653262215

I think it is a similar problem but I was unable to solve it by replacing it with the part of the code they recommend, I get other errors that I can't understand.

MKS-Sean, 2020-07-04:
@nictronik99 It seems to report the error that the library file does not exist.

nictronik99, 2020-07-04:
@MKS-Sean I used the file downloaded from here: https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware
Without making any changes, I tried to compile it as it is

Miraculix200, 2020-07-05:
What's the version of your STM32 platform? If it's version 7.0.0 then it's probably caused by updating the platform.

nictronik99, 2020-07-05:
> What's the version of your STM32 platform? If it's version 7.0.0 then it's probably caused by updating the platform.

@Miraculix200  yes it is

PR #47 (2020-07-10): Update platformio.ini
fix #44 error described here: https://github.com/MarlinFirmware/Marlin/pull/18516
commit 772436be (2020-07-10): Add files via upload

fix #44 error described here: https://github.com/MarlinFirmware/Marlin/pull/18516
closed 2020-07-10 by makerbase-mks