makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #45: Touchscreen sensitivity

swalsh11, 2020-07-04:
I can get this to Build in VS code based of of the latest Bugfix release on Marlin, for a Twotrees sapphire plus, Core XY,  however somthing is not correct, from the HMI, I can get the axis to jog in the correct directions most of the time, but it behaves as if the touch screen is picking up phantom touches, 
Homing is a disaster, I cannot get it to home at all in XY, and Z goes the wrong way, most of the home screen appears dead.

The Screen works Perfectly in the mkS firmware 2.031

All basic setting copied from this working config: 
https://github.com/swalsh11/Marlin-2.0.X-MKS-Robin-Nano

My Setting: 
https://github.com/swalsh11/Marlin/tree/Basic-Sapphire-Plus-Config
 
I would like to start looking at the screen sensitivity,  how do I get a debug output for my finger press location?

makerbase-mks, 2020-07-11:
@swalsh11 You can try to update to the latest version to test it.

swalsh11, 2020-07-12:
Ok, I will recompile from the latest bugfix

mks-viva, 2020-12-11:
I will close this issues temporarily.

closed 2020-12-11 by mks-viva