makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #48: Problem Firmware origina

grau0, 2020-07-05:
Hello everyone, I just got the nano v1.2 + TFT 35 + TMC2209, and I have a problem with allí original firmware, the motors only rotate in one direction, even though I try through the LCD it does not rotate (if reversing the rotation in the LCD rotates  the other way, but not both ways).  
If I load marlin, it works fine, but the icons on the LCD come out without text. 
Can somebody help me?  
Thank you.

makerbase-mks, 2020-07-11:
`Hello everyone, I just got the nano v1.2 + TFT 35 + TMC2209, and I have a problem with allí original firmware, the motors only rotate in one direction, even though I try through the LCD it does not rotate (if reversing the rotation in the LCD rotates the other way, but not both ways).`

Are you using marlin firmware? Is the option of TMC2209 enabled in the firmware?

`If I load marlin, it works fine, but the icons on the LCD come out without text.`

Can you provide an image file on this issue?

mks-viva, 2020-12-11:
If there is no response, then I will temporarily close this issues.

closed 2020-12-11 by mks-viva