makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #49: Stuck after Updating Firmware

dgorbunov, 2020-07-06:
Copied everything from the Firmware folder onto TF card and put into MKS Robin Nano, turned on, TFT shows updating, and it goes through updating the font, but after is just stuck on a black screen. I've waited minutes to see if anything happened to no avail. It's stuck on booting if I try to plug in again.

dgorbunov, 2020-07-06:
Issue resolved, used a different TF/microSD card

closed 2020-07-06 by dgorbunov

PeteyPub, 2021-08-02:
I'm having a similar issue. My NEPTUNE hangs on Font Table Updating 100%.
I left it on over night thinking maybe its just slow, no luck. I have tried 6 different size and brand SD cards, from 2GB to 16GB, formatted to FAT32, NTFS, FAT.
NO LUCK
I've also tried different versions of the FIRMWARE. Nothing. If I skip the FONT update it hangs at a different spot

poon-holder, 2022-03-21:
I had this issue with a Robin nano V1.1 and 1.3. Both just stuck at "100%" then displayed only "booting" after reboot without SD.  The issue seem temperamental, hard to replicate after success and non dependant on SD card (having tried many types as originally thought would be the case. 
It is VERY misleading if it is the first time you are setting up klipper as the .bin file changes to .cur extension giving the illusion the flash is successful, plus you do not know what the screen should do (nothing) after successful klipper flash. 
Any one more successful at diagnosing the cause of this issue? My smooth brain cannot delve deeper im afraid.