makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #50: how to set double stepper driver for Z axis

nictronik99, 2020-07-10:
hi, how can i set double stepper driver for Z axis using 2nd stepper on E1 connector?

makerbase-mks, 2020-07-11:
In the configuratio_adv.h file you need to define:
#define NUM_Z_STEPPER_DRIVERS 2

nictronik99, 2020-07-14:
> In the configuratio_adv.h file you need to define:
> #define NUM_Z_STEPPER_DRIVERS 2

@makerbase-mks but how does Marlin know that i Will attach the 2nd Z motor to E1 connector ??

makerbase-mks, 2020-07-15:
`#if NUM_Z_STEPPER_DRIVERS >= 2
  #ifndef Z2_STEP_PIN
    #define Z2_STEP_PIN   _EPIN(Z2_E_INDEX, STEP)
    #define Z2_DIR_PIN    _EPIN(Z2_E_INDEX, DIR)
    #define Z2_ENABLE_PIN _EPIN(Z2_E_INDEX, ENABLE)
    #if Z2_E_INDEX >= MAX_EXTRUDERS || !PIN_EXISTS(Z2_STEP)
      #error "No E stepper plug left for Z2!"
    #endif
  #endif
#endif
`
@nictronik99  The code Will redefine the pin of Z2.

nictronik99, 2020-07-15:
> `#if NUM_Z_STEPPER_DRIVERS >= 2 #ifndef Z2_STEP_PIN #define Z2_STEP_PIN _EPIN(Z2_E_INDEX, STEP) #define Z2_DIR_PIN _EPIN(Z2_E_INDEX, DIR) #define Z2_ENABLE_PIN _EPIN(Z2_E_INDEX, ENABLE) #if [...]

perfect, thanks a lot

adrianbelano, 2021-02-19:
> `#if NUM_Z_STEPPER_DRIVERS >= 2 #ifndef Z2_STEP_PIN #define Z2_STEP_PIN _EPIN(Z2_E_INDEX, STEP) #define Z2_DIR_PIN _EPIN(Z2_E_INDEX, DIR) #define Z2_ENABLE_PIN _EPIN(Z2_E_INDEX, ENABLE) #if [...]

So I need to type this to the configuration_adv.h ?

closed 2021-02-21 by MKS-Sean