makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #51: New icons

Ranil007, 2020-07-13:
Hi mks! I need the bin icons wich is on the photo in the folder "images". Can you send it me please 🙏

makerbase-mks, 2020-07-14:
@Ranil007 The style icon bin file you want has been updated at this link.
https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/tree/master/Firmware

Ranil007, 2020-07-14:
> @ Ranil007 Файл бина значка стиля, который вы хотите, был обновлен по этой ссылке.
> https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/tree/master/Firmware

I install it, by I don't have those icons(((

Ranil007, 2020-07-14:
> @Ranil007 The style icon bin file you want has been updated at this link.
> https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/tree/master/Firmware

I install it, by I don't have those icons(((

makerbase-mks, 2020-07-18:
How is the screen displayed? Can you take a picture and show me

closed 2021-02-21 by MKS-Sean