makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #52: WIFi module and marlin

nictronik99, 2020-07-15:
Hi I write to know if the wifi module provided in Robin nano 1.2 can be used with this version of Marlin and if so how do you configure it?

makerbase-mks, 2020-07-18:
I will add this feature recently.

themacboy, 2020-07-26:
now is up

nictronik99, 2020-07-28:
> now is up

@themacboy can you  link some guide or maybe explain how to set this ?

themacboy, 2020-07-29:
> > now is up
> 
> @themacboy can you link some guide or maybe explain how to set this ?

there is no extensive documentation about it. Check the wifi repository for more information.

Actualy I only use it with prontface or cura. Trying to find a way to make web interface more usefull.

sobieh, 2020-08-17:
Currently it's kind of useless as it simply doesn't work.

Where can i find these files btw ?
MksWifi.bin
MksWifi_Web.bin
MksWifi_WebView.bin

themacboy, 2020-08-18:
> 
> 
> Currently it's kind of useless as it simply doesn't work.
> 
> Where can i find these files btw ?
> MksWifi.bin
> MksWifi_Web.bin
> MksWifi_WebView.bin

MksWifi.bin is here:

https://github.com/makerbase-mks/MKS-WIFI/blob/master/firmware_release/MksWifi_CS1.0.4_200227.zip

Other files dint know if exist anywhere