makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #53: Build targeting mks_robin_nano35_v2 does not work

mgcrea, 2020-07-17:
I received an MKS Robin Nano V2.0 from AliExpress that I had to reflash to configure a delta printer (FLSUN Q5).

I received an [MKS Robin Nano V2.0 from AliExpress](https://www.aliexpress.com/item/4001185795620.html) that I had to reflash to configure a delta printer (FLSUN Q5).

Did reflash with the latest [Mks_Nano35_V2.0.3.3 Release file.rar](https://github.com/makerbase-mks/MKS-Robin-Nano/tree/master/Nano%20firmware%20v2.x.x)

Unfortunately the printer does not work anymore as I have a blank white screen after boot.

So I did successfully compile this code with env=`mks_robin_nano35_v2` and flashed using the generated build in `.pio/build`, but I now have a black screen!

Any idea what I could be missing?

Thanks!

nictronik99, 2020-07-17:
1) Follow the steps in the photos. 
2) compiled firmaware robin_nano35.bin is in the directory .pio / build / mks_robin_nano35_v2 (warning: the file you're looking for Is Robin_nano35.bin not firmware.bin)

[image]
[image]
[image]

mgcrea, 2020-07-17:
Thanks, it does start OK, however it looks the GUI is far from being ready, Marlin GUI features are not available as `TFT_LITTLE_VGL_UI` does seem to be a separate beast with only minor customization... For instance I can't set `DELTA_CALIBRATION_MENU` as it complains about a missing `HAS_GRAPHICAL_LCD`.

nictronik99, 2020-07-18:
> Thanks, it does start OK,
> 
> Unfortunately I have trouble configuring the AutoLevel on an FLSun Q5, basically applied [these Marlin configuration patches](https://github.com/AndersSahlman/Marlin/compare/07509febcd0e4d9ad0083426faaffd113e1473ee...AndersSahlman:Flsun-Q5-2208) but the printer probes a single point and stops.
> 
> Also it looks like the serial communication somehow hangs (can't send GCODE).
> 
> Looks like I have put a bit too much hope in this new V2 board as it does not seem to be ready for prime time yet!
> 
> Thanks for the help anyway.

you are absolutely right, I too, like you, was expecting something else and instead I find myself with an absolutely immature card without documents ...

mgcrea, 2020-07-18:
Yep I'm a bit disappointed by the UI so far as it feels like a regression migrating from the V1 (except bigger screen). But I guess it will be better with time.

Anyway good news, finally successfully calibrated my delta through the USB serial connection (basically G33 followed by M665 HXXX.XX to adjust Z offset) and I got a great test print from it with dimensions way more accurate than with the legacy firmware that came with the printer.

So pretty happy finally!

trashypopashy, 2020-07-21:
Did anyone manage it, to get the Robin Nano V2 to work with the included display in the old 80s style?

ils15, 2020-07-26:
> Yep I'm a bit disappointed by the UI so far as it feels like a regression migrating from the V1 (except bigger screen). But I guess it will be better with time.
>
> Anyway good news, finally [...]

Hi mgcrea

V2.0 still under developing using Marlin features
old mks robin (V1.2) compiled under marlin 1.0 and using a proprietary compilator
Now we are under testing new features and merging it into original marlin repository
but, every menu, every button, every function need to be converted do LVGL, tested, before marlin bugfix commits
you can follow discussion in these PR
https://github.com/MarlinFirmware/Marlin/pull/18500
https://github.com/MarlinFirmware/Marlin/pull/18784

closed 2021-02-21 by MKS-Sean