makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #54: MKS Robin Nano V2.0 -- robin_nano35_cfg.txt is not run

HulusiAKAR, 2020-07-22:
Hello!
I bought the product 2 days ago. (MKS Robin Nano V2.0 +  TS35 v2.0 + MKS Wifi + 5xTMC2209).
My 3d printer => Anet A8
My 3d printer work dual z step motor and bltouch.
I could not find the installation scheme of this version on the Internet.
Moreover, robin_nano35_cfg.txt file does not work.
How do I install Bltouch, wifi, dual z step motor and tmc2209 uart.
I would appreciate if you make a detailed explanation.
Thank you so much.

makerbase-mks, 2020-07-23:
Hello, are you familiar with the use of marlin? The firmware of nano_v2.0 is designed based on marlin, and the usage is the same as the marlin firmware. You need to enable the functions you want in configuration.h, such as TM2209, BL_TOUCH functions

mks-viva, 2020-11-19:
I will close this issues temporarily.

closed 2020-11-19 by mks-viva