makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #55: How to add servo42a in marlin ...

themacboy, 2020-07-23:
Hello, just purchased 2 MKS-SERVO42A, but I'm not sure if they need any special config to be used in Robina Nano v1.2 and Robin Nano v.2 (I have both)

I'm not skilled in Marlin and Saint Google is not helping me a lot.

Can you explain how to do it in the config file.

Thx in advance !

makerbase-mks, 2020-07-28:
https://github.com/makerbase-mks/MKS-SERVO42A
Can this help you?

themacboy, 2020-07-29:
Not really.

What I request is about the parameters I must set in congiguration.h in marlin file before compiling.

For example, thasta correct?

**#define Z_DRIVER_TYPE  A4954LPT**

makerbase-mks, 2020-07-29:
You don’t need to configure this item, you can use the default parameters normally

closed 2020-10-18 by themacboy