makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #57: Question, is there a config example for 256 native micro step with 2208 Uart?

TheSin-, 2020-07-24:
I have been trying to get my 2208s in 256 native microsteps but I get serious noise/jerking on fast movements, is there an example of it?

makerbase-mks, 2020-07-28:
Have you configured uart mode?

TheSin-, 2020-07-28:
it is in uart, it works perfectly in 16 native emulating 256.  I adjusting all the steps through up the config, my best guess is when it does quick movements it must be changing modes and one of the mods on the 2208 is the issue.  I was going to try to lock in a mode so see if that helps but I haven't had time.  Hence I thought I'd ask if there is an example config for 256 native as I haven't been able to find one online yet.  It's that or I'm missing something someplace.

makerbase-mks, 2020-07-29:
#define HYBRID_THRESHOLD
Maybe you can define this option in the advanced configuration, when your speed is too high, switch to spreadCycle mode

TheSin-, 2020-07-29:
Thanks I'll read up on that one and give it a try.  In the mean time I just ordered an SGEN-L V2 with TMC2209s maybe that'll work out better for me ;)  I love MKS stuff so it's so hard to decide.

closed 2021-02-21 by MKS-Sean