makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #59: Support for MKS Robin Nano V2.0

HulusiAKAR, 2020-07-27:
There are many documents and videos for MKS robin nano v1.2 on the internet, but there are no documents and videos for robin nano v2.0. We have difficulty in installing. For mks robin nano v2.0 we want editable robin_nano35_cfg.txt. If I knew this would happen, I would get v1.2 instead of Mks robin nano v2.0.

makerbase-mks, 2020-07-28:
https://github.com/makerbase-mks/MKS-Robin-Nano/wiki/MKS-Robin-Nano-V2.0-User-Manual
Can this document help you?

HulusiAKAR, 2020-09-10:
NO !

Noferix, 2020-09-13:
Me ocurre lo mismo, al tratar de compilar el marlin y actualizar el firmware se queda colgado en tft updating 100% y ya no inicia, tambien he tratado de configurar los ajustes con mkstool version web pero no controlo nada de la impresora, los endstop no funcionan, el motor de z gira al revés, y tampoco consigo configurar doble motor en z, es todo un desastre, he tenido que comprar la versión 1.2 con el archivo nano35.txt que es muchísimo mas simple y rápido. Se esta trabajando en una solución para todo esto? o tengo que deshacerme de esta placa?

makerbase-mks, 2020-09-14:
@HulusiAKAR @Noferix The last version firmware has more partment can be change and setting, you can update it.

anlkaplan, 2020-09-15:
@HulusiAKAR Merhaba motorların jumperlarını takamadımda şuana kadar sizin kartın fotoğrafını atabilir misiniz

Noferix, 2020-09-15:
@HulusiAKAR Los drivers encajan perfectamente, mi placa es una robin nano 2 y los drivers tmc2225, aunque la placa cuenta con pines adicionales que en los drivers no estan soldados y no se para que son.

themacboy, 2020-09-15:
Prueba a borrar todo lo que hay dentgro de la carpeta .pio para que fuerce
a bajar todas las librerias desde zero y compila de nuevo.

Tambien puedes probar a primero restaurar con una versión pre compilada (la
de la carpeta firmware) si se recupera entonces poner la tuya. A mi esto
tabmien me ha solucionado algun problema.

Missatge de Noferix <notifications@github.com> del dia dg., 13 de set. 2020
a les 12:06:

themacboy, 2020-09-15:
this documnet and the wiki are very useful !

But more video example can help a lot.

Missatge de makerbase <notifications@github.com> del dia dt., 28 de jul.
2020 a les 13:55:

Noferix, 2020-09-15:
@themacboy Fue lo primero que pense, "restablecer de fabrica el firmware" y volver a compilar marlin de 0 pero aun asi me ha dado problemas y no se ni por que, de hecho intente modificar solo el tipo de drivers y en cuando compilo ya me ocurre este problema y no entiendo nada. He revisado la wiki y buscado en youtube y en internet pero la placa es tan nueva que no se tiene casi ninguna informacion :(

themacboy, 2020-09-15:
la version pre-compilada tampoco te carga, la de la carpeta firmware que
compila mks?

Missatge de Noferix <notifications@github.com> del dia dt., 15 de set. 2020
a les 18:23:

Noferix, 2020-09-15:
La original si, es la que utilizo para "restaurar" pero cuando compilo una nueva versión solo modificando los drivers se queda clavado al 100% y al reiniciar la placa aparece boting... hasta que vuelto a "restaurarla"

meteozond, 2020-11-24:
@Noferix I've ran into this problem too. Looks like you are trying to flash wrong file: `.pio/build/mks_robin_nano35/firmware.bin` instead of `.pio/build/mks_robin_nano35/Robin_nano35.bin`

mks-viva, 2021-09-02:
The latest version firmware has support it. So, i will close it

closed 2021-09-02 by mks-viva