makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #60: Help for Bltouch setup and settings for MKS Robin Nano v2.0

HulusiAKAR, 2020-07-27:
I could not adjust bltouch settings for MKS robin nano v2.0. Can you help me?

makerbase-mks, 2020-07-28:
What trouble have you encountered?
The use of BL_TOUCH requires corresponding configuration on configuration.h

maalouc, 2020-08-10:
Same Here! 3d touch connected to PA8 Servo0, and PC4 for ZMAX Endstop, no response for deploy on command M280 P0 S10.
All the configs done according to the Nano V2 Wiki.

SENDING:M115
FIRMWARE_NAME:Marlin 2.0.6 (Aug 10 2020 00:15:39) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Tornado V1.1 EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Cap:SERIAL_XON_XOFF:0
Cap:BINARY_FILE_TRANSFER:0
Cap:EEPROM:1
Cap:VOLUMETRIC:1
Cap:AUTOREPORT_TEMP:1
Cap:PROGRESS:0
Cap:PRINT_JOB:1
Cap:AUTOLEVEL:1
Cap:RUNOUT:0
Cap:Z_PROBE:1
Cap:LEVELING_DATA:1
Cap:BUILD_PERCENT:0
Cap:SOFTWARE_POWER:0
Cap:TOGGLE_LIGHTS:0
Cap:CASE_LIGHT_BRIGHTNESS:0
Cap:EMERGENCY_PARSER:0
Cap:PROMPT_SUPPORT:0
Cap:SDCARD:1
Cap:AUTOREPORT_SD_STATUS:0
Cap:LONG_FILENAME:0
Cap:THERMAL_PROTECTION:1
Cap:MOTION_MODES:0
Cap:ARCS:1
Cap:BABYSTEPPING:0
Cap:CHAMBER_TEMPERATURE:0

>>> M119
SENDING:M119
Reporting endstop status
x_min: open
y_min: open
z_min: open
z_max: open
z_probe: open

because no deploy on G29 drops probe error.

>>> G29
SENDING:G29
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
Error:Probing Failed
[ERROR] Error:Probing Failed

On machine power On 3Dtouch perform the deploy and stow diagnostic. but not working on g29.

maalouc, 2020-08-10:
update: the config option: 
#define Z_MIN_PROBE_PIN PC4 // Pin 32 is the RAMPS default
it's not working, m119 show not triggered status for the probe, but show triggered for the zmax pin where it's the probe connected.

btw im using zmax, because zmin it's used for hardware endstop.

>>> M119
SENDING:M119
Reporting endstop status
x_min: open
y_min: open
z_min: open
z_max: TRIGGERED
z_probe: open

i try the same on marlin bugfix2 and it's work properly, but want to find the solution on mks firmware.

maalouc, 2020-08-10:
more info:

MKS Firmware (Not Working)
>>> M43 S1
SENDING:M43 S1
Servo probe test
. using index:  0, deploy angle: 10, stow angle:   90
. Probe Z_MIN_PROBE_PIN: 36
. Z_MIN_PROBE_ENDSTOP_INVERTING: false
. Check for BLTOUCH
. Deploy & stow 4 times
** Please trigger probe within 30 sec **
FAIL: No trigger detected

>>> M43
SENDING:M43
PIN:   0  PA0  (A 0)  HEATER_BED_PIN                         protected
PIN:   1  PA1  (A 1)  E1_DIR_PIN                              TIM2 CH2
PIN:   2  PA2  (A 2)  POWER_LOSS_PIN                         Input  = 1 TIM2 CH3
PIN:   3  PA3  (A 3)  E1_ENABLE_PIN                           TIM2 CH4
PIN:   4  PA4  (A 4)  SDSS                                   Input  = 1
.                     SS_PIN                                 Input  = 1
PIN:   5  PA5  (A 5)  SCK_PIN                                Output = 0
.                     TOUCH_SCK_PIN                          Output = 0
PIN:   6  PA6  (A 6)  MISO_PIN                               Input  = 1 TIM3 CH1
.                     TOUCH_MISO_PIN                         Input  = 1 TIM3 CH1
PIN:   7  PA7  (A 7)  MOSI_PIN                                TIM3 CH2
.                     TOUCH_MOSI_PIN                          TIM3 CH2
PIN:   8  PA8         SERVO0_PIN                              TIM1 CH1
PIN:   9  PA9         <unused/unknown>                        TIM1 CH2
PIN:  10 PA10         <unused/unknown>                        TIM1 CH3
PIN:  11 PA11         Z_MIN_PIN                              protected
.                     Z_DIAG_PIN                             protected
.                     Z_STOP_PIN                             protected
PIN:  12 PA12         Y_MIN_PIN                              protected
.                     Y_DIAG_PIN                             protected
.                     Y_STOP_PIN                             protected
PIN:  13 PA13         <unused/unknown>                       Input  = 1
PIN:  14 PA14         <unused/unknown>                       Input  = 0
PIN:  15 PA15         X_MIN_PIN                              protected
.                     X_DIAG_PIN                             protected
.                     X_STOP_PIN                             protected
PIN:  16  PB0  (A 8)  E0_AUTO_FAN_PIN                        protected
.                     HEATER_1_PIN                           protected
PIN:  17  PB1  (A 9)  FAN_PIN                                protected
PIN:  18  PB2         <unused/unknown>                       Input  = 0
PIN:  19  PB3         E0_ENABLE_PIN                          protected
PIN:  20  PB4         Z_DIR_PIN                              protected
PIN:  21  PB5         Z_STEP_PIN                             protected
PIN:  22  PB6         <unused/unknown>                        TIM4 CH1
PIN:  23  PB7         <unused/unknown>                        TIM4 CH2
PIN:  24  PB8         Z_ENABLE_PIN                           protected
PIN:  25  PB9         Y_DIR_PIN                              protected
PIN:  26 PB10         <unused/unknown>                       Output = 0
PIN:  27 PB11         <unused/unknown>                       Input  = 1
PIN:  28 PB12         <unused/unknown>                       Output = 1
PIN:  29 PB13         <unused/unknown>                       Output = 0
PIN:  30 PB14         <unused/unknown>                       Input  = 1
PIN:  31 PB15         <unused/unknown>                       Output = 0
PIN:  32  PC0  (A10)  <unused/unknown>   Analog in =     0   Input  = 0
PIN:  33  PC1  (A11)  <unused/unknown>   Analog in =     0   Input  = 0
PIN:  34  PC2  (A12)  <unused/unknown>                       Input  = 1
PIN:  35  PC3  (A13)  HEATER_0_PIN                           protected
PIN:  36  PC4  (A14)  Z_MIN_PROBE_PIN                        protected
.                     E0_DIAG_PIN                            protected
PIN:  37  PC5  (A15)  BEEPER_PIN                             Output = 0
PIN:  38  PC6         <unused/unknown>                        TIM8 CH1
PIN:  39  PC7         <unused/unknown>                        TIM8 CH2
PIN:  40  PC8         <unused/unknown>                        TIM8 CH3
PIN:  41  PC9         <unused/unknown>                        TIM8 CH4
[... 136 lines cut ...]
.                     E1_SERIAL_RX_PIN                       Input  = 0
PIN:  57  PD9         E0_SERIAL_TX_PIN                       Output = 1
.                     E0_SERIAL_RX_PIN                       Output = 1
PIN:  58 PD10         <unused/unknown>                       Output = 1
PIN:  59 PD11         <unused/unknown>                       Output = 1
PIN:  60 PD12         <unused/unknown>                       Input  = 0
PIN:  61 PD13         LCD_BACKLIGHT_PIN                      Output = 1
PIN:  62 PD14         <unused/unknown>                       Input  = 1
PIN:  63 PD15         E1_STEP_PIN                            Input  = 1
PIN:  64  PE0         Y_STEP_PIN                             protected
PIN:  65  PE1         Y_ENABLE_PIN                           protected
PIN:  66  PE2         X_DIR_PIN                              protected
PIN:  67  PE3         X_STEP_PIN                             protected
PIN:  68  PE4         X_ENABLE_PIN                           protected
PIN:  69  PE5         <unused/unknown>                       Input  = 1
PIN:  70  PE6         FIL_RUNOUT2_PIN                        Input  = 1
PIN:  71  PE7         E1_DIAG_PIN                            Input  = 1
[... cut ...]
PIN:  73  PE9         <unused/unknown>                       Input  = 1
PIN:  74 PE10         <unused/unknown>                       Input  = 1
PIN:  75 PE11         BTN_EN2                                Input  = 1
PIN:  76 PE12         <unused/unknown>                       Input  = 1
PIN:  77 PE13         BTN_ENC                                Input  = 1
PIN:  78 PE14         TOUCH_CS_PIN                           Output = 1
PIN:  79 PE15         <unused/unknown>                       Input  = 1

maalouc, 2020-08-12:
problem solve after last commit!

closed 2021-02-21 by MKS-Sean