makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #61: [BUG] Build with board type BOARD_MKS_ROBIN_NANO does not compile: 'WIFI_IO1_PIN' was not declared in this scope

yousry, 2020-07-29:
Step to reproduce this problem:     
Download the sourcecode Version 2.0.5.3 via the MKS-TOOL or use the configuration from this repository.

Bug Description after 'clean' and 'build':

```
In file included from Marlin/src/lcd/extui/lib/mks_ui/src/../../../../../inc/../HAL/./STM32F1/HAL.h:36:0,
                 from Marlin/src/lcd/extui/lib/mks_ui/src/../../../../../inc/../HAL/HAL.h:26,
                 from Marlin/src/lcd/extui/lib/mks_ui/src/../../../../../inc/MarlinConfig.h:30,
                 from Marlin/src/lcd/extui/lib/mks_ui/src/../../../../../MarlinCore.h:24,
                 from Marlin/src/lcd/extui/lib/mks_ui/src/irq_overrid.cpp:1:
Marlin/src/lcd/extui/lib/mks_ui/src/irq_overrid.cpp: In function 'void __irq_usart1()':
Marlin/src/lcd/extui/lib/mks_ui/src/../inc/../inc/wifi_module.h:37:31: error: 'WIFI_IO1_PIN' was not declared in this scope
 #define WIFI_IO1_SET()  WRITE(WIFI_IO1_PIN, HIGH);//GPIO_SetBits(GPIOC, GPIO_Pin_7)
```

makerbase-mks, 2020-07-29:
Does the code downloaded from mks-tool not compile normally? How can you download the code on github to compile normally?

makerbase-mks, 2020-07-29:
Need to define pin in pins_MKS_ROBIN_NANO.h file：
#define WIFI_IO1_PIN       			        PC7
#define WIFI_RESET_PIN			              	PA5
I have modified it in the latest submission

yousry, 2020-07-29:
Commit 6bdb52679b4bb7e9a7071f2f8f63bf2086843e62 fixed the problem. Thank you!

closed 2020-07-29 by yousry