makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #62: black screen after upload latest firmware to board

jeor112, 2020-07-29:
Hi,
I´m looking for an alternativ firmware than the MKS stock firmware for my MKS_robin_nano V1.2 and MKS Robin TFT35 V1.0.
Reason why is, that I have to configure switching nozzle and BLTouch.

When I upload this version, Mks-Robin-Nano-Marlin2.0-Firmware, I end up with a black screen.
I havn´t compiled this in Vs Code PIO, just uploaded the bin file from the firmware folder.

Please, what am I doing wrong?

I have an older version MKS Marlin2 firmware configured for switching nozzle and BLTouch, but there is an issue with the BLTouch 9 point mesh. The probing failes after 6 points.
I havn´t been able to sort this issue. I have been recommended to update the Marlin firmware, but that just leaves me with much more worse problems.

Thanks

makerbase-mks, 2020-07-29:
The assets in the Firmware folder are robin_nano2.0 firmware
You need to compile the firmware of robin_nano1.2 after configuration：
default_envs = mks_robin_nano35
#define MOTHERBOARD BOARD_MKS_ROBIN_NANO
#define TFT_LVGL_UI_FSMC  // Robin nano v1.2 uses FSMC

jeor112, 2020-07-29:
Thank you very much, that did it!
Then I have an additional question.

Will this menu support the two extruders and the switching nozzle configuration or do I have to configure this for the menu as well? I use a digital servo for switching nozzle with signal on pin PC7. (BLTouch on pin PA8)
If so, can you please easily explain me how to add these functions to the menu?

Thank you very much in advance!

jeor112, 2020-07-30:
Have compiled for the Nano v.1.2 board as you have described and at first enabled the BLTouch, following the guid in the Nano V2 Wiki https://github.com/makerbase-mks/MKS-Robin-Nano-V2/wiki/3D_TOUCH
I use Z_MIN_PROBE_PIN (pin PA11) and not Z_MAX_PROBE_PIN (pin PC4) as described in the guid:
"**#define Z_MIN_PROBE_PIN PC4  //32 // Pin 32 is the RAMPS default**"

Then I got the followng issues:
1. When homing the axises, then the BLTouch dosn´t deploy.
2. When executing the Auto bedleveling, the X axsis doesn´t move smoothly to center of the bed.
When the probe reaches the center position, the probe still doesn´t deploy and starts moving to the bed.

Another issue I have:

How do I disable the Wifi option?
I need to use the IO1 pin (PC7) for a second servo for the switching nozzle.
As I understand, using the IO1 PC7 pin is the only option on the Nano V1.2 board. 
When I uncomment the Wifi IO pins in the pins_MKS_ROBIN_NANO.h file and instead define the SERVO1_PIN to PC7 pin, I got errors when comping.
I can´t figure out what else to uncomment by looking at the compiling errors.

Sorry for my ignorance.

Thanks

makerbase-mks, 2020-07-31:
>

jeor112, 2020-08-01:
Hi

Thanks for your reply!

**`How do I disable the Wifi option?
#define USE_WIFI_FUNCTION 0

You can configure it to 0 in the wifiSerial.h file`**

Well, that didn´t help. Still not able to compile.

Sorry I doesn´t have the error after define USE_WIFI_FUNCTION 0, because I now have started all over step by step.
Status for now:

Homing do not work, so I can´t go further.

I have attatched a zip file with my config files.
Below, I have write down the changes I have made in the configuration.h:

[Conf settings.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/5010036/Conf.settings.zip)

#define EXTRUDERS 2

#define USE_XMIN_PLUG
#define USE_YMIN_PLUG
#define USE_ZMIN_PLUG
//#define USE_XMAX_PLUG
//#define USE_YMAX_PLUG
//#define USE_ZMAX_PLUG

// Enable pullup for all endstops to prevent a floating state
#define ENDSTOPPULLUPS
#if DISABLED(ENDSTOPPULLUPS)
  // Disable ENDSTOPPULLUPS to set pullups individually
  //#define ENDSTOPPULLUP_XMAX
  //#define ENDSTOPPULLUP_YMAX
  //#define ENDSTOPPULLUP_ZMAX
  //#define ENDSTOPPULLUP_XMIN
  //#define ENDSTOPPULLUP_YMIN
  //#define ENDSTOPPULLUP_ZMIN
  //#define ENDSTOPPULLUP_ZMIN_PROBE
#endif

// Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup).
#define X_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Y_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Z_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define X_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Y_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Z_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Z_MIN_PROBE_ENDSTOP_INVERTING false // Set to true to invert the logic of the probe.

#define DISTINCT_E_FACTORS

/**
 * Default Axis Steps Per Unit (steps/mm)
 * Override with M92
 *                                      X, Y, Z, E0 [, E1[, E2...]]
 */
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 400, 400, 400 }

/**
 * Default Max Feed Rate (mm/s)
 * Override with M203
 *                                      X, Y, Z, E0 [, E1[, E2...]]
 */
#define DEFAULT_MAX_FEEDRATE          { 300, 300, 5, 25, 25 }

//#define LIMITED_MAX_FR_EDITING        // Limit edit via M203 or LCD to DEFAULT_MAX_FEEDRATE * 2
#if ENABLED(LIMITED_MAX_FR_EDITING)
[... 139 lines cut ...]
 */
#define RESTORE_LEVELING_AFTER_G28

#define Z_SAFE_HOMING

#if ENABLED(Z_SAFE_HOMING)
  #define Z_SAFE_HOMING_X_POINT X_CENTER  // X point for Z homing
  #define Z_SAFE_HOMING_Y_POINT Y_CENTER  // Y point for Z homing
#endif

// Homing speeds (mm/m)
#define HOMING_FEEDRATE_XY (50*60)
#define HOMING_FEEDRATE_Z  (4*60)

// Validate that endstops are triggered on homing moves
#define VALIDATE_HOMING_ENDSTOPS

#define TFT_LVGL_UI_FSMC  // Robin nano v1.2 uses FSMC

#define NUM_SERVOS 1 // Servo index starts with 0 for M280 command

// (ms) Delay  before the next move will start, to give the servo time to reach its target angle.
// 300ms is a good value but you can try less delay.
// If the servo can't reach the requested position, increase it.
#define SERVO_DELAY { 300 }

jeor112, 2020-08-01:
I forgot to mention, that I have also tried the Marlin 2.0.6 and tried to set it up with Marlin UI, but that ended up with black screen no matter what I tried.
I starting to have the feeling, that something is not right with the version 2.0.6 related to MKS Robin Nano version 1.2 and TFT35 version 1.0

The best configuration I have had so fare is the Marlin Bugfix version 2.0.0 with the Marlin UI.
The only issue with that version is, that the 9 point auto bedleveling stops at the 6 step and leveling failes.
I haven´t been able to sort that issue out

makerbase-mks, 2020-08-01:
Are you using BL_TOUCH or 3DTOUCH? I will test it when I go to work.

jeor112, 2020-08-01:
Thanks
Oh yes, it´s a 3DTOUCH
Are there any difference between these two?
If so, I wasn´t aware of that. :-(

jeor112, 2020-08-04:
Hi
I have made a video showing the issue with the X moving when it homes and bedleveling.
Remember to listen to the sound.
[Homing problems Listen to the sound.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/5020907/Homing.problems.Listen.to.the.sound.zip)

jeor112, 2020-08-04:
As I mentioned earlier, I have best luck with the Marlin 2.0.0 Bugfix.
The only issue I got with this one is, that it stops leveling at the 6 point out of 9 points.
Here is a copy af the Marlin folder if you want to look into this.
[Marlin Bugfix_V2.0.0_MksRobinNano.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/5020991/Marlin.Bugfix_V2.0.0_MksRobinNano.zip)

If you want to upload the configuration, then remember to rename the Robin_nano.bin to Robin_nano35.bin

jeor112, 2020-08-07:
All fixed by using the official Marlin 2.0.6 with Marlin UI

closed 2020-08-07 by jeor112