makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #66: SD card read fails while updating display

MB-Finski, 2020-08-04:
My printer is based on the MKS Robin Nano version used by Lotmaxx (they have their own silk screen on the PCB but I think it's v1.1 of the board and has a 3.5" TFT over FSCM).

The firmware uploaded without any issues and I even managed to get some good prints done over USB. However, printing from the SD card fails as soon as the TFT display is refreshed (typically the first major update after homing causes a fail). I think the read operation from the SD-card is compromized by the simultaneous update of the display. Maybe simultaneous reads from the sd-card (g-code and menu images) causing the issue? DMA related issue?

I tried slowing down the SD card spi interface to no avail. Printing from SD card can be done if it is initiated over USB (no display updates). 

This *might* be a hardware related issue as the Lotmaxx original firmware has a noticeable pause in printing if the screen updated during a print (like when going through menus) so I think the original firmware blocks g-code reads from the SD card while updating the display. 

Has anyone else here experienced this bug? Any easy workarounds?

MB-Finski, 2020-08-04:
Btw, I uploaded the "factory version" 2.0.3.4 MKS firmware and it doesn't have this same issue so I think I can confirm this is a bug (or misconfiguration of my config files).

doon3, 2020-08-11:
Hey. I have the same problem. MKS Robin Nano 1.2v. Have you defeated this problem?

doon3, 2020-08-12:
Partially solved this problem: the gcode files must be placed directly in the SD cards, not in a folder.

MB-Finski, 2020-08-12:
> Partially solved this problem: the gcode files must be placed directly in the SD cards, not in a folder.

For me, printing directly from the SD card root doesn't really help (I haven't even tried printing from anywhere else. There's something funky going on with the SPI communications and switching from one device to another (SD-card reader to flash memory) screws up the read operations from the SD card.

MB-Finski, 2020-09-02:
Just following up on this issue: Compiled the latest available source (2 days ago) and I can confirm that it still has this issue. 

Printing stops sometimes before even the bed is heated. Sometimes the first line is printed. I tried enabling debugging messages over USB but it gave no error/debug messages. Occasionally the message "echo 'unrecognized command "05"" is printed over USB when this error happens but not every time.

MB-Finski, 2020-09-03:
If anyone is reading these:

Some further updates on trying to narrow down the possible source of the issue. I disabled the LVGL GUI and used "FSMC_GRAPHICAL_TFT" with  instead. That seemed to solve the issue (of course, now the graphics won't get loaded from the external flash all the time). So I think that narrows it down to the flash reads(/writes?) causing issues with the SD-card communication.

forbin2575, 2020-11-17:
I recently bought the Sainsmart Coreception 300 which also uses the MKS Robin Nano V1.2 board.  It works fine with the dated standard firmware (which seems to be from the Creativity Elf), but I wanted to be able to use Marlin and its updated features.  I was able to get Marlin built for the Robin board and TFT but when I print a model the printer pauses at the first G1 gcode command.  I did some debugging and found its because the software thinks the filament has run out.  I think this may also be the cause of some of the pauses described above.  I was able to remove the filament from the filament sensor and then Marlin printed properly (with no other changes to the printer or the gcode).  You could give that a try to see if you have the same problem I had.  If so, here are the changes I made to the Marlin source configuration to fix it:

    In the file Marlin/src/pins/stm32f1/pins_MKS_ROBIN_NANO.h
        Change "#define MT_DET_2_PIN                      PE6" to
                     "//#define MT_DET_2_PIN                      PE6"
        Change "#define MT_DET_PIN_INVERTING        false" to
                     "#define MT_DET_PIN_INVERTING        true"

My printer only has one filament sensor (MT_DET_1_pin) so commenting out the define disables testing the empty MT_DET_2 sensor connections.  Changing the detection assertion state provides correct filament detection function.  Other printers using the MKS Robin Nano V1.2 may also need this change.  Hope this helps.

Malderin, 2020-11-17:
> I recently bought the Sainsmart Coreception 300 which also uses the MKS Robin Nano V1.2 board. It works fine with the dated standard firmware (which seems to be from the Creativity Elf), but I [...]

Yes it works. Thanks.
For mks robin anno v.2 board I did this:
1. > In the file Marlin/src/pins/stm32f1/pins_MKS_ROBIN_NANO_v2.h
>     Change "#define MT_DET_2_PIN                      PE6" to
>                  "//#define MT_DET_2_PIN                      PE6"
>     Change "#define MT_DET_PIN_INVERTING        false" to
>                  "#define MT_DET_PIN_INVERTING        true"
2. In configuration.h
//#define FILAMENT_RUNOUT_SENSOR
#define FILAMENT_RUNOUT_SENSOR
//  #define FILAMENT_RUNOUT_SCRIPT "M600"
#define FILAMENT_RUNOUT_SCRIPT "M25"

MB-Finski, 2020-11-22:
Yes, I can confirm these changes fix the issue. Is the line #define FIL_RUNOUT_PIN PA4 necessary, though?

forbin2575, 2020-11-23:
I left the //#define FILAMENT_RUNOUT_SENSOR option commented out, but the filament runout sensor works without that option enabled.  Makes me wonder how.  Maybe a defualt runout sensor logic turned off when you choose the FILAMENT_RUNOUT_SENSOR option.   I was able to comment out the pins_MKS_ROBIN_NANO.H "//#define FIL_RUNOUT_PIN PA4" and it compiled successfully, so FIL_RUNOUT_PIN must be defined somewhere else.   I just upgraded to a BTT SKR V1.4 Turbo board with the BTT TFT35 screen, so I can't quickly switch back to test your configuration.

X-Ryl669, 2020-11-26:
@Malderin, on my system, it works out of the box, it won't print without filament. I had to rewire the connector through, but I think it's an issue with the manufacturer (Two Trees) who inverted the connection in there (they wired it initially so that it was open whatever a filament was present or not, so it was 100% useless).
I do have FIL_RUNOUT_PIN defined to PA4 even if it's also defined in the STM32's header, because it was failing to compile at that time.

mdunfee3016, 2022-07-06:
> MT_DET_PIN_INVERTING

This was my issue, to the point i disconnected the filament out sensor bypass board and printing resumed. Now to fix the bed size issue with the BLTouch