makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #75: Image problems in TS35 Robin Nano V2

maalouc, 2020-08-12:
incorrect display of filament load/unload and wifi settings.

[image]
[image]

After compile copied the bin file and assets folder to SD.

BTW filament unload/load function working perfect!

doon3, 2020-08-12:
you need to correctly insert the files into the folder mks_font and mks_pic... copy this folders from \Mks-Robin-Nano-Marlin2.0-Firmware-master\Firmware... to SD card

maalouc, 2020-08-12:
Thanks that worked!

closed 2020-08-12 by maalouc