makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #76: Robin Nano V2 Wifi problem

maalouc, 2020-08-12:
sending to print from mks cura plugin to the board over wifi display hang on message Uploading.... after reboot the board the file its on the sdcard but size 0.

[image]

thanks in advance!

makerbase-mks, 2020-08-13:
Hi! I am sorry! This firmware version does not currently support file transfer. We will add the function of transferring gcode file by WIfi In the future.

maalouc, 2020-08-13:
thanks! will close the issue then! waiting for the feature :)

closed 2020-08-13 by maalouc

themacboy, 2020-08-13:
And cloud? Or is not supported yet? I can't see the cloud icon

HulusiAKAR, 2020-09-10:
It has been 2 months since I bought the device and we are still waiting for an update. 
When will the device receive a Wifi update? We cannot use the wifi plug.
I have an error every time. We wouldn't buy this device if I knew that would happen.
Hear our voices now.