makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #80: Robin Nano v1.2 problems:

doon3, 2020-08-16:
uncommented:
#define MOTHERBOARD BOARD_MKS_ROBIN_NANO
#define TFT_LVGL_UI_FSMC 
...
problems:
1. works badly pause/resume when printing.. Z returning to ptrinting earlier XY, 
does not work retract on pause.
2. does not work retract on end of print.
3. percentage is not displayed on the print progres screen
4. printing is work only from the first directory of the SD card (... SD card/file.gcode) . If files are in the folder (... SD card/folder/file.gcode) - the printer does not print
5. number keyboard - have you tried to enter a number using it ?? it's almost unrealistic ...

doon3, 2020-08-31:
in the new version fixed the problem 5. - number keyboard work good. Other problems remain.... noticed a new problem - babystep does not show Z offset, always shows 0.00...

protektwar, 2021-12-22:
Hello,

I have a MKS Robin Nano V1.2 TFT35 Board for which I would like to compile this Marlin 2.0 firmware.

I had problems compiling the code as specified in MKS Robin Nano V1.x build and update firmware section.
I'm using Ubuntu with Visual Studio and PlatformIO.

Only after I solved some problems as follows:
I had to: (in this order)
 1. add to platformio.ini:
   - in section[env:mks_robin_nano35]
       platform: ststm32@<6.2.0
   - in section [common]
       lib_deps: Wire and Adafruit BusIO
 2. add to SPI.h
       uint8_t transfer(uint8_t *buf, size_t count) const;
 3. add to SPI.cpp
    void SPIClass::transfer(uint8_t *buf, uint16_t count) {
       transferBytes(reinterpret_cast<uint8_t*>(buf), nullptr, count);
    }

Now that the build for mks_robin_nano35 was finished with success, I copied the following to the SD Card:
 -  from Firmware Folder: 
           - mks_font
           - mks_pic
 - from .pio/build/mks_robin_nano35/
           - Robin_nano35.bin
Update was ok, "Booting..." was written on Display but then nothing else happen, it's black and it's not doing anything.
I have tried also with this option(#define TFT_LVGL_UI_FSMC), but it's the same.

I really need help with this build as I have problems with board, as I'm using external drivers SS570 and these are not working at all, as the timing between ENA, DIR and STEP, als the Pulse STEP width is not ok. The Drivers work ok with my CNC board controlled over LTP with MACH3.

Thanks!