makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #85: Robin Nano v1.2

doon3, 2020-08-29:
Hello.  after update does not work Robin Nano v1.2. - the screen does not display icons

TimCarrDP, 2020-08-30:
Try including the assets folder on the root of the SD when updating, this is where the fonts and images are stored.

doon3, 2020-08-30:
thanks, I did it ... this problem occurred after an update ... with earlier versions the images were updated ...

makerbase-mks, 2020-08-31:
@doon3 So your black screen problem has been solved?

doon3, 2020-08-31:
Hello . the screen turns on, but there are no icons ...

doon3, 2020-08-31:
solved this problem. bin files were placed not in two folders (mks_font, mks_pic), but in one (assets)

diver197, 2020-08-31:
Mine, too, is not displayed. There is a text display of updated resources and a running screen

doon3, 2020-08-31:
put all bin files in one folder (assets) to sd directory....

diver197, 2020-09-01:
I compared all the files, changed the case, compile and download is still not displayed

MKS-Sean, 2021-02-21:
@diver197 Is your screen still unable to display?

closed 2021-09-02 by mks-viva