makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #87: Weird layer inconsistencies

TimCarrDP, 2020-08-30:
Comparing the same (hand translated) Config files from le3tspeaks port to this one, the MKS version has weird layer inconsistencies.

MKS version:
[image]

Le3tspeak version:
[image]

makerbase-mks, 2020-08-31:
I have to find some time to test it. You can upload the gcode file. Are you using nano 1.2?

TimCarrDP, 2020-08-31:
Nano 1.2, yes.
Gcode is simple test http://www.mediafire.com/file/ldrq5to6xu4xan3/file

TimCarrDP, 2020-09-03:
@makerbase-mks any chance to see this? I hate to say it but if the print quality is significantly lower than with other marlin firmware, it makes this port useless.
There has been some nice work with the LCD so it would be a shame!

makerbase-mks, 2020-09-03:
@TimCarrDP You provide the configuration.h and configuration_adv.h files, we will test it tomorrow.

TimCarrDP, 2020-09-04:
Sorry been out on work and not had access to my config files.
Here they are
http://www.mediafire.com/folder/g3u1xzckc6fl0

TimCarrDP, 2020-09-06:
@makerbase-mks did you manage to recreate the issue?

makerbase-mks, 2020-09-10:
Yes, I am still waiting for testers to test it.

TimCarrDP, 2020-09-17:
Do you need a hand testing @makerbase-mks @MKS-Sean?
Sadly this port is useless until fixed!