makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #88: Robin Nano v1.2 problems:

doon3, 2020-08-31:
uncommented:
#define MOTHERBOARD BOARD_MKS_ROBIN_NANO
#define TFT_LVGL_UI_FSMC

In the new version keyboard work good. Other problems that I wrote about earlier have not been fixed....
Noticed a new problem - babystep does not show Z offset, always shows 0.00...

JinwooJang, 2020-08-31:
uncomment in configuration_adv.h

#define BABYSTEP_DISPLAY_TOTAL
#define BABYSTEP_ZPROBE_OFFSET

doon3, 2020-08-31:
JinwooJang did not help...
I uncoment : #define BABYSTEP_DISPLAY_TOTAL
//#define BABYSTEP_ZPROBE_OFFSET - I cannot uncomment, errors appear...
but, Z offset it works for me ... only it always shows 0.000

makerbase-mks, 2020-09-03:
#define BABYSTEPPING
#define BABYSTEP_ZPROBE_OFFSET
Enable these two configuration items in the latest firmware, it can work normally

mks-viva, 2020-11-19:
I will close this issues temporarily.

closed 2020-11-19 by mks-viva