makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #89: Bluer Plus TFT43

wookie76, 2020-09-02:
Greetings:

I can compile and flash, but I can't get this to work with my new TwoTrees Bluer Plus featuring a Robin Nano and TFT43.  

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

TimCarrDP, 2020-09-03:
Any errors? What's the symptoms?

wookie76, 2020-09-04:
Flash appears to work, then TFT43 goes white.  In looking around, the TFT43 appears to be too new.

makerbase-mks, 2020-09-05:
@wookie76  I think it’s not updated to the correct firmware, right?

mks-viva, 2020-12-11:
If there is no response, then I will temporarily close this issues

closed 2020-12-11 by mks-viva