makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #90: TFT35 Flip Orientation

dgorbunov, 2020-09-03:
Is there an option to flip the orientation of the TFT screen 180 degrees in Marlin? I'm running an MKS Robin Nano v1.2 with TFT35 v1.0. 

I haven't found it in Configuration.h, Configuration_adv.h, or pins_MKS_ROBIN_NANO.h. This should seriously be added if it doesn't exist.

makerbase-mks, 2020-09-04:
I put it in this file: tft_lvgl_configuration.h
#define GRAPHICAL_TFT_ROTATE_180

dgorbunov, 2020-09-05:
> I put it in this file: tft_lvgl_configuration.h
> #define GRAPHICAL_TFT_ROTATE_180

This does not work how it should. It literally flips the x-axis of the screen making the text unreadable. It needs to flip the y-axis as well. It is not rotating the screen 180, it is flipping all text and pictures over the x axis

makerbase-mks, 2020-09-10:
Is the flipped interface abnormal? I have fixed the touch problem

Sekelofbabel, 2020-11-09:
I am having the same issue.

itisnot-me, 2020-11-18:
@makerbase-mks This issue still persists. I need to get this corrected as the manufacture decided to start switching things around and im either stuck with bad firmware or an upside down screen. 

see this image to show what it is doing. 
https://i.imgur.com/EQBP1Ro.jpg

I looked at the code but since I do not know how to change this one part I an unable to fix this myself.

Karlostavitch1, 2021-02-09:
+1 having the same issue here too.

Gitdad, 2021-02-20:
+1 having the same issue.

Gitdad, 2021-02-20:
Just figured it out. the TFT_ROTATION needs to be in tft_io.h instead of tft_lvgl_configuration.h

in tft_io.h it looks like this.

// TFT_ROTATION is user configurable
#ifndef TFT_ROTATION
  #define TFT_ROTATION TFT_ROTATE_180
#endif

infinisean, 2021-08-24:
I just put 2.0.7.2 on a new Sapphire Plus I got and this issue is still there.
I configured #define TFT_ROTATION TFT_ROTATE_180
so now the interface looks correct, but the actual press-areas are catty-corner to where they appear to be.
The recommendation from Gitdad above looks to already be in place in this version to no effect.
Is there another workaround for this?

jerhole, 2021-09-19:
> I just put 2.0.7.2 on a new Sapphire Plus I got and this issue is still there.
> I configured #define TFT_ROTATION TFT_ROTATE_180
> 

+1

chiluk, 2021-12-20:
I'm hitting this same issue where the I can fix the display, but the touch screen also needs to be rotated 180.  Anyone figure out how to rotate the touchscreen 180 as well?

chiluk, 2021-12-21:
I discovered that if during calibration I tap the opposite corner of the target I get a semi-functional display.  Basically when it says top left, instead tap bottom right.  I'm thinking I may give up on the LVGL gui and move to the marlin gui instead.  How bad is it with only a touch screen?

jerhole, 2021-12-22:
> <makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/issues/90/998468563@
> github.com>

I'll have to give that a try. It does make sense that would flip the
orientation. I'm using Marlins UI, definitely not convinced to lvlgui

hmsq82, 2021-12-28:
I'm experiencing an issue, when I flashed this on my Robin Nano V2 TS35 V2.0 the touch screen is not correct it is  flipped horizontally.

So if I touch top right corner, it accepts touch on bottom left corner of screen.

hmsq82, 2022-01-06:
I just ended up doing a compile directly from MarlinFW 2.0.9.3 and it is working fine. So not sure what is wrong with MKS marlin version.

chiluk, 2022-01-06:
@hmsq82 any chance you can push up a repo with your configs?  I have a sapphire plus, and I'd love to collaborate with someone on keeping a semi-up-to-date functional config around for it.

itisnot-me, 2022-01-06:
If you guys can compile, it's actually really easy, you can use this setup and change anything you need to. https://github.com/DJScopeSOFM/Two-Trees-Multi-Printer-Marlin-Bugfix-Compiler it is a little older than the most recent but it's what I'm using on two of my printers right now

datarius2, 2022-01-06:
I would say that you need to make a separate config for the version of sapphire plus v1.1

datarius2, 2022-01-06:
there the display is turned 180 and the touchscreen in the rack when you rotate the display remains

jaberkwt, 2022-01-25:
> > Hi after installing marlin on my mks nano 1.2 the bootloader got flipped now I returnd to the stock firmware of two trees blueR plus and the screen still flipped in the bootloader and the interface any help?
> 
> got the same issue. there is a mks35screen firmware that can be edited to rotate the screen, but ive been unable to get that to work. it flashes but bootloader keeps being upside down as all the text. can rotate screen as described above with new marlin flash. But unfortunately the touchscreen is still scuffed then. also got a weird grey bar ontop the screen that goes away when reflashing the orginal firmware.
> 
> board "MKS robin nano 1.2" with "TFT3.5screen" that has a ribbon cable and no sd slot. eg: "sapphire plus"

Thank alot Frank I contact the technical support they gave me the original firmware after flashing it twice everything work fine with original firmware but no luck with marlin.

hmsq82, 2022-01-25:
> @hmsq82 any chance you can push up a repo with your configs? I have a sapphire plus, and I'd love to collaborate with someone on keeping a semi-up-to-date functional config around for it.

I actually compiled it for another machine TT-1s, but I have compiled it  also for my Sapphire Plus, but it's on my spare board, let me swap out with the one on the printer and if it all works and I will push it to my git. I'm just waiting on some replacement parts for the Sapphire Plus before I do any of this.

jamesjnadeau, 2022-05-13:
I have a fresh two trees sapphire plus(now called SP5) and I was able to get this working with:
```
#define TFT_ROTATION TFT_ROTATE_180
```
Hope this helps the next guy :)

ejhoness, 2022-05-21:
upload firmware with assets folder in memory card

ton4eff, 2022-09-30:
I'm trying to make Marlin 2.1 to work with Robin nano v1.3 (SP-5 - Sapphire Plus v2.1). The screen rotates properly with TFT_ROTATE_180, but the touchscreen does not, if using LVGL_UI. Everything works properly with classic and color ui... So is there a way to rotate touchscreen too for the mks tft35?

jamesjnadeau, 2022-10-02:
@ton4eff, after you install, you need connect to the board via the serial terminal via pronterface, or octoprint, etc.

Then you need to run this to run the screen calibration:

https://marlinfw.org/docs/gcode/M995.html

ton4eff, 2022-10-02:
@jamesjnadeau, thanks, I have to press exactly the opposite x spots on the touchscreen in order to make it work properly. It works, but I wonder if there is a way to make it right by default in config marlin files...