makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #91: Filament Runout Detection?

dgorbunov, 2020-09-04:
How do you enable use of MT_DET_PIN for filament detection? FIL_RUNOUT_PIN cannot be used because it requires a standard LCD to work. How does filament detection work with MT_DET_PIN?

TimCarrDP, 2020-09-04:
It does work actually. Don't have it in front of me, but you just need to change the M600 command to M25. I'll find it and get back to you, but just search for the M600 command in the mean time

TimCarrDP, 2020-09-04:
Its line   #define FILAMENT_RUNOUT_SCRIPT "M600"

Change it too   #define FILAMENT_RUNOUT_SCRIPT "M25"
and it should compile and work.

dgorbunov, 2020-09-06:
And you replace MT_DET_PIN  with FIL_RUNOUT_PIN, correct?

TimCarrDP, 2020-09-06:
Nope. The LCD is no longer required if you use M25

makerbase-mks, 2020-09-10:
@dgorbunov If you use the lvgl ui interface, you will use MT_DET_PIN.

AdamMulica, 2020-10-30:
Hello, i search this 
#define FIL_RUNOUT_PIN     PA4   // MT_DET - that is, a white socket on the motherboard with 3 pins
#define NUM_RUNOUT_SENSORS   1  or 0 - on/off ?

I have detection sensor with 2pins (closed with filament) and i must connect to S G ? - no 5V

On the display I have the option: exstrusion and I need to have min. 170 degrees to use this option? I would like to check a stepper motor. How to use this option without enabling hotend ?

misiek271, 2020-11-24:
I change to M25 script. And its working in half. :) printing is stop when filament is gone. but extruder dont unload filament. Then you change filament manualy. and click reasume and Extruder start Unloading filament ant then printer starts to print. :) so something is wrong with that script.

X-Ryl669, 2020-11-25:
I can confirm. If you click resume, it should either ask to change filament (unload, pause with asking to confirm it's done, and load a new filament), or to resume directly. The same happens if you click "pause" on the printing page, upon clicking resume, it unloads the filament too!

misiek271, 2020-11-25:
So it works good in your case? Coz i dont have any info on screen that about filament.

X-Ryl669, 2020-11-26:
No it doesn't work, and it doesn't work exactly like you said. Right now, if you click "Pause" on the printing page, it lift up the hotend (good), and waits here. It would be cool if it retracted the filament a bit, since it flows, but anyway.
Once you press "Resume", it unload the filament (why ?) and then, without any filament, tries to continue printing.
It should:
1. Ask if you pause for changing filament (or you can change screen and unload and load filament yourself)
2. Resume without unloading the filament
But that's not what it does.

misiek271, 2020-11-26:
So its behavior like on this movies?? Coz mine do the same https://drive.google.com/folderview?id=1AMdf7lnrnXHmyWKkkqzWjOtDjpUtSiR5

mks-viva, 2020-11-26:
@X-Ryl669 @misiek271 
Hi! We are testing and verifying this.

X-Ryl669, 2020-11-26:
@misiek271 Yes it does the same, except that I have a Bowden so it unloads 90cm of filament for me (hence it's impossible for it to resume printing without so many filament missing).

Malderin, 2020-11-26:
I was able to start the filamet runout sensor. Look here:
https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/issues/66#issuecomment-729179983

X-Ryl669, 2020-11-26:
@Malderin Thanks. On my system, I don't need anything for the filament sensor detection to work (in fact, I'm almost sure it ignores the enable ifdef anyway). My configuration is [here](https://github.com/X-Ryl669/Mks-Robin-Nano-Marlin2.0-Firmware-SapphirePlus/tree/mine). It seems you are compiling the latest Marlin firmware (2.0.7.2). Did you include latest MKS's change in your version or is it a vanilly Marlin's clone with your configuration ?

Malderin, 2020-11-26:
> @Malderin Thanks. On my system, I don't need anything for the filament sensor detection to work (in fact, I'm almost sure it ignores the enable ifdef anyway). My configuration is [here](https://github.com/X-Ryl669/Mks-Robin-Nano-Marlin2.0-Firmware-SapphirePlus/tree/mine). It seems you are compiling the latest Marlin firmware (2.0.7.2). Did you include latest MKS's change in your version or is it a vanilly Marlin's clone with your printer configuration ?

My version (bltогch, wifi and filament runout sensor) is here:
https://github.com/Malderin/Robin-nano-V2-kingroonKP3/tree/KingRoonKP3_test

misiek271, 2020-11-26:
no he's not. he didnt uncoment lvlg screen so i presume that hes using TFT not TS. i checked his config.

vzjrz, 2022-05-08:
So if I'm reading this right, there is no current way to use the 'BIGTREETECH Smart Filament Sensor' with this firmware and TFT_LVGL_UI?

DKiefer770, 2022-11-29:
Is there anyone who has a MARLIN configuration for the 3.0 version of this board that will allow the filament runout and Advanced Pause/Filament change to work correctly?  I have tried setting FILAMENT_RUNOUT_SCRIPT "M600" and it works somewhat: the head parks, beeps, unloads filament, reloads filament, then just stays at "preheating extruder" even though it reaches the correct temperature.  

Similar problem with FILAMENT_RUNOUT_SCRIPT "M25", the head parks, then moves back to the center of the print area, then just stops at "parking print head".  Adding FILAMENT_LOAD_UNLOAD_GCODES doesn't appear to do anything.

/**
 * Advanced Pause for Filament Change
 *  - Adds the G-code M600 Filament Change to initiate a filament change.
 *  - This feature is required for the default FILAMENT_RUNOUT_SCRIPT.
 *
 * Requirements:
 *  - For Filament Change parking enable and configure NOZZLE_PARK_FEATURE.
 *  - For user interaction enable an LCD display, HOST_PROMPT_SUPPORT, or EMERGENCY_PARSER.
 *
 * Enable PARK_HEAD_ON_PAUSE to add the G-code M125 Pause and Park.
 */
#define ADVANCED_PAUSE_FEATURE
#if ENABLED(ADVANCED_PAUSE_FEATURE)
  #define PAUSE_PARK_RETRACT_FEEDRATE         60  // (mm/s) Initial retract feedrate.
  #define PAUSE_PARK_RETRACT_LENGTH           20  // (mm) Initial retract.
                                                  // This short retract is done immediately, before parking the nozzle.
  #define FILAMENT_CHANGE_UNLOAD_FEEDRATE     10  // (mm/s) Unload filament feedrate. This can be pretty fast.
  #define FILAMENT_CHANGE_UNLOAD_ACCEL        25  // (mm/s^2) Lower acceleration may allow a faster feedrate.
  #define FILAMENT_CHANGE_UNLOAD_LENGTH      100  // (mm) The length of filament for a complete unload.
                                                  //   For Bowden, the full length of the tube and nozzle.
                                                  //   For direct drive, the full length of the nozzle.
                                                  //   Set to 0 for manual unloading.
  #define FILAMENT_CHANGE_SLOW_LOAD_FEEDRATE   6  // (mm/s) Slow move when starting load.
  #define FILAMENT_CHANGE_SLOW_LOAD_LENGTH    10  // (mm) Slow length, to allow time to insert material.
                                                  // 0 to disable start loading and skip to fast load only
  #define FILAMENT_CHANGE_FAST_LOAD_FEEDRATE   6  // (mm/s) Load filament feedrate. This can be pretty fast.
  #define FILAMENT_CHANGE_FAST_LOAD_ACCEL     25  // (mm/s^2) Lower acceleration may allow a faster feedrate.
  #define FILAMENT_CHANGE_FAST_LOAD_LENGTH    10  // (mm) Load length of filament, from extruder gear to nozzle.
                                                  //   For Bowden, the full length of the tube and nozzle.
                                                  //   For direct drive, the full length of the nozzle.
  //#define ADVANCED_PAUSE_CONTINUOUS_PURGE       // Purge continuously up to the purge length until interrupted.
  #define ADVANCED_PAUSE_PURGE_FEEDRATE        3  // (mm/s) Extrude feedrate (after loading). Should be slower than load feedrate.
  #define ADVANCED_PAUSE_PURGE_LENGTH         20  // (mm) Length to extrude after loading.
                                                  //   Set to 0 for manual extrusion.
                                                  //   Filament can be extruded repeatedly from the Filament Change menu
                                                  //   until extrusion is consistent, and to purge old filament.
  #define ADVANCED_PAUSE_RESUME_PRIME          0  // (mm) Extra distance to prime nozzle after returning from park.
  //#define ADVANCED_PAUSE_FANS_PAUSE             // Turn off print-cooling fans while the machine is paused.

                                                  // Filament Unload does a Retract, Delay, and Purge first:
  #define FILAMENT_UNLOAD_PURGE_RETRACT       13  // (mm) Unload initial retract length.
  #define FILAMENT_UNLOAD_PURGE_DELAY        500  // (ms) Delay for the filament to cool after retract.
  #define FILAMENT_UNLOAD_PURGE_LENGTH         8  // (mm) An unretract is done, then this length is purged.
  #define FILAMENT_UNLOAD_PURGE_FEEDRATE      25  // (mm/s) feedrate to purge before unload

  #define PAUSE_PARK_NOZZLE_TIMEOUT          180  // (seconds) Time limit before the nozzle is turned off for safety.
  #define FILAMENT_CHANGE_ALERT_BEEPS          5  // Number of alert beeps to play when a response is needed.
  #define PAUSE_PARK_NO_STEPPER_TIMEOUT           // Enable for XYZ steppers to stay powered on during filament change.
  //#define FILAMENT_CHANGE_RESUME_ON_INSERT      // Automatically continue / load filament when runout sensor is triggered again.
  //#define PAUSE_REHEAT_FAST_RESUME              // Reduce number of waits by not prompting again post-timeout before continuing.

  #define PARK_HEAD_ON_PAUSE                      // Park the nozzle during pause and filament change.
  //#define HOME_BEFORE_FILAMENT_CHANGE           // If needed, home before parking for filament change

  #define FILAMENT_LOAD_UNLOAD_GCODES           // Add M701/M702 Load/Unload G-codes, plus Load/Unload in the LCD Prepare menu.
  //#define FILAMENT_UNLOAD_ALL_EXTRUDERS         // Allow M702 to unload all extruders above a minimum target temp (as set by M302)
#endif