makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #93: No icons ...

themacboy, 2020-09-09:
Good morning,

after update to last versión, and used assets folder, no icon is showed.

You only see lottlet white squares with letters (No data).

I have tryed to the old method using MKS folder, and surprise, same problem.

I have lost all icons, in any way.

Any know problem or solution?

themacboy, 2020-09-09:
[image]

themacboy, 2020-09-09:
[image]

makerbase-mks, 2020-09-10:
Hello, I think your asset file version is too old, you need to download the latest one, if the same problem persists. You can create an assets folder yourself, then download the files under the following path, copy the bin files in mks _pic and mks_font to the assets folder, and then copy them to the SD card to update
https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/tree/master/Firmware

themacboy, 2020-09-10:
I use a direct fork from your github, with all your last commits merged in
my repository.

themacboy, 2020-09-10:
I have updated with the old precompiled firmware that I found in:

...\.pio\libdeps\mks-assets.zip

Then all icons are showed again.

After that I have updated again with new version ... and icon disapear again.

Have you made and important change between the old version and the current version?

themacboy, 2020-09-12:
At the moment I have disconnected the MKS UI, and turned on the Marlin UI.

themacboy, 2020-09-14:
> Hello, I think your asset file version is too old, you need to download the latest one, if the same problem persists. You can create an assets folder yourself, then download the files under the [...]

Now I have understand you ! And it is working !!! Thx

closed 2020-09-14 by themacboy

themacboy, 2020-09-14:
The simplest way to solve that problem is delete "mks-assets.zip" file in the ".../.pio/libdeps/... " fodler